betaflight/betaflight-configurator #595: 3.2.0 Blackbox logging device settings failing to store, works in CLI

swilkens, 2017-08-04:
On version 3.2.0 of the Chrome Store version of the configurator, the blackbox logging device configuration has moved to the blackbox tab. This dropdown to select the logging device doesn't seem to actually store the setting after clicking save & reboot.

Steps to reproduce:
1. Clean flash BF 3.2.0-RC2
2. Connect using BFC 3.2.0
3. In the blackbox tab, notice the device is set to SDCARD
4. Set the device to "No Logging" and click save and reboot
5. Notice the setting didn't take and is still set to SDCARD
6. Through CLI, check the current setting with get blackbox_device and notice it is still set to SDCARD

Setting the blackbox_device through the CLI does work and the configurator does update to show that I changed it to NONE through the CLI. But updating this through the configurator does nothing.

swilkens, 2017-08-04:
Since a recent pull request mentions this should be fixed in the FC FW now: https://github.com/betaflight/betaflight/pull/3715

I tried to flash the most recent build (betaflight_3.2.0_BETAFLIGHTF3-50-ac03d2983.hex) of the firmware but that didn't fix it either. I still can't set Blackbox logging device to no logging while on betaflight_3.2.0_BETAFLIGHTF3-50-ac03d2983.hex

DieHertz, 2017-08-04:
The food was on configurator's side, not firmware.

closed 2017-08-04 by DieHertz