betaflight/betaflight-configurator #597: bad message for configurationDisarmKillSwitch

piepluquad, 2017-08-05:
Hello guys,

if I understand it well, the message in configuration tab -> ESC/Motor Features that is

`"Disarm motors regardless of throttle value (When arming via AUX channel)"`

should be : 

`"Disarm motors regardless of throttle value (When **dis**arming via AUX channel)"`

or tell me if i miss something ?

happy flying 👍

Arakon, 2017-08-05:
I understood the text to refer to the arming **function** via aux, not the specific act of arming.

piepluquad, 2017-08-06:
configurationDisarmKillSwitch corresponds to disarmkillswitch which corresponds to ARMING_CONFIG.disarm_kill_switch returned by armingConfig()->disarm_kill_switch(), 
so now looking better at `betaflight-master\src\main\fc\rc_controls.c`:164 it seems we are talking about when the  `disarm()` function is called :

```
// perform actions
if (!isUsingSticksToArm) { // pieplu'HO : arming&disarming via AUX channel, not stick input
    if (IS_RC_MODE_ACTIVE(BOXARM)) { //  pieplu'HO : arm switch is in ARM position
        rcDisarmTicks = 0;
        // Arming via ARM BOX
        tryArm(); //  pieplu'HO : try to arm 
    } else { // pieplu'HO :  arm switch is in DISARM position
        // Disarming via ARM BOX
        resetArmingDisabled();

        if (ARMING_FLAG(ARMED) && rxIsReceivingSignal() && !failsafeIsActive()  ) {//  pieplu'HO :  
// the quad is already armed and receiving inputs from TX 
// and failsafe is not active => AUX DISARMED switch is activated by user 
// (but perhaps we don't want the quad to drop from the air if the switch switched by accident,
// so we check configuration choosen by user to prevent unwanted crash:
            rcDisarmTicks++;
            if (rcDisarmTicks > 3) {
                if (armingConfig()->disarm_kill_switch) { // pieplu'HO : user wants to kill quad when DISARM switch is choosen
                    disarm();
                } else if (throttleStatus == THROTTLE_LOW) { //pieplu'HO  only kill quad if throttle is at ZERO
                    disarm();
                }
            }
        }
    }
}
```
I added in that code some  comments with `pieplu'HO` and now i'm pretty sure that the message for configurationDisarmKillSwitch shall be : 
"Disarm motors regardless of throttle value (When disarming via AUX channel)"
or : 
"Disarm motors during flight (if throttle value is not LOW when disarming via AUX channel => killswitch)"

I was first surprised as a quad beginner to crash into ground with throttle not at zero, trying to disarm quad with switch, and quad was trying to dig a hole. So now i understand how to use it : be sure to have throttle at min AND disarm switch to disarm quad, OR : activate the DisarmKillSwitch to disarm 'regardless of throttle value.

thanks for reading
what do you think now?

piepluquad, 2017-08-12:
comment : my taranis won't let me arm if throttle not LOW

piepluquad, 2017-08-16:
perhaps message could be : "Arm or Disarm motors regardless of throttle value (When using AUX channel Arm BOX)"

mikeller, 2017-08-17:
@piepluquad: This is definitely wrong. The switch only influences the behaviour when disarming. Arming is still disabled when the throttle is not low.

piepluquad, 2017-08-18:
thanks mikeller for involving in this thread, i agree with you
just the original message ""Disarm motors regardless of throttle value (When arming via AUX channel)"" confused me at first, better change it to be less confusing i guess...

mikeller, 2017-08-18:
@piepluquad: Would you like to open a pull request for this change?

piepluquad, 2017-08-18:
ok, i'll try it, i'm kinda able to do this,
what do you think of this message instead : 
`Arm/Disarm motors regardless of throttle value (When using AUX channel)`

piepluquad, 2017-08-18:
i tried to create a pull request, but i think i did it wrong (doesn't look like others pull requests... 
can i fix it, and how ?

mikeller, 2017-08-19:
The format of the pull request is good, well done. But the new message that you are proposing needs to be amended.

piepluquad, 2017-08-19:
ok, i understand better now, 
so in fact the sentence
Disarm motors regardless of throttle value (When arming via AUX channel)
is good, i just didn't understood the behaviour.
Perhaps telling the user that
> Arming is disabled when the throttle is not low.

could be a good info here in help of the configuration option ?

piepluquad, 2017-08-19:
sorry to make loose your time, i'm so noobish.
i was thinking : perhap something like this : 
`Disarm motors regardless of throttle value (When arming / disarming is configured in Modes tab with AUX channel)` ?

closed 2017-08-20 by mikeller