betaflight/betaflight-configurator #601: Disable all OSD elements at once

spuder, 2017-08-09:
The OSD currently has 37 elements that can be configured. If a user wants to only use a handful of elements, they need to do a lot of clicking to disable all of them. 

As a user, I want to press a button to clear all active elements, so I can select just the elements I want. 

I envision something like this

[image]

[image]

piepluquad, 2017-08-09:
kinda duplicate of #560 ?

spuder, 2017-08-09:
Yes, that is a duplicate.

closed 2017-08-09 by spuder