betaflight/betaflight-configurator #603: Motor PWM Speed Separated from PID Speed with DSHOT

richard-scott, 2017-08-10:
The documentation on [DSHOT ](https://github.com/betaflight/betaflight/wiki/DSHOT-ESC-Protocol) states:

> When DSHOT is enabled Unsyced PWM is disabled. DSHOT always runs at the PID loop rate.

However, when setting DSHOT as your ESC protocol in the Configuration Tab the option to separate PWM speed from PID speed is still there.

Is that correct? I would assume this option would vanish?

SpoonFPV, 2017-08-27:
I've checked you can click it and it doesn't do anything on Dshot, The ESCs are still synced with the PID loop.  I would consider this a non-issue, other than the option should be grayed out if it doesn't do anything.

richard-scott, 2017-09-07:
That's exactly my point and why I raised the issue :-)

Other features seem to vanish from the GUI when they are not compatible with the options selected. For example, going from Multishot to Dshot removes:

- Minimum Throttle
- Maximum Throttle
- Minimum Command

Likewise in other areas activating LED_STRIP introduces a new menu and removes the menu item when you deactivate the slider. It would be nice to keep consistency throughout the GUI.

closed 2017-10-15 by richard-scott