betaflight/betaflight-configurator #604: 3.2.1 OSD Fonts fail to upload

1xGSEadamx, 2017-08-11:
Configurator Version: 3.2.1
Board: DYSF4Pro
Release: 3.2 RC3

**Steps to recreate:**
Click OSD Tab
Click Font Manager
Click Digital Font
Click Upload
**Expected Result**: Font should upload success and be reflected in the OSD
**Actual Result:** Upload completes and board disconnects. After reconnecting the board the OSD reflects the old font.

jflyper, 2017-08-11:
@1xGSEadamx Not reproducible. All Fonts uploads and displayed without a problem.

[image]

Default
[image]

Bold
[image]

Large
[image]

Extra Large
[image]

Betaflight
[image]

Digital
[image]

1xGSEadamx, 2017-08-11:
@jflyper Strange - when I upload the font, the board reboots after the progress bar completes but the change is not saved. Also revisiting the OSD tab in betaflight shows the old font.

Arakon, 2017-08-12:
Works here on 3 boards too. Did you actually check the font change on screen? The preview screen does NOT properly reflect the active font.
Also, make sure your OSD is getting enough power through your USB port.. some standalone OSDs had trouble with the font upload if they didn't get a stable voltage.

1xGSEadamx, 2017-08-14:
Retested and the font is reflected on the actual OSD but the preview is not updating. Updating Issue details

Arakon, 2017-08-14:
Problem with that is, that there is no way for the configurator to actually know what font is loaded on the OSD.

basdelfos, 2017-08-14:
The configurator and BF don't know which font is uploaded. Maybe good to
add a message to inform the shown font in de configurator OSD preview is no
an actual reflection of the installed font.

tigert, 2017-08-17:
A workaround (when I worked on my own font) is to upload it, then reconnect and go to font uploader, select the font and then cancel the upload. A bit cumbersome, but the font gets loaded into configurator, and at least you can see your font on the virtual OSD without needing a quad with all the features working.

I guess it could be solved with an "uploaded font" setting in betaflight, which would be a bit sketchy since it would not really be a real setting, rather than a stored piece of information. I guess you could store the path to the last font you uploaded and reload it when you connect a board, which of course would work only on the same computer you uploaded the font with.

closed 2018-05-13 by stale[bot]