betaflight/betaflight-configurator #606: 3.2.1 Blackbox-log download broken

Grwmagad, 2017-08-12:
Hey, 

I am using a Kakute F4 on BF 3.2 rc2.
When I try to download the blackbox log from the internal flash the pop up opens and immediatly says completed. 
Obviously the file is not working with any blackbox explorer since its empty. 

Workaround is to use an older version of the betaflight configurator. 3.1.3 is working for example. Same fc, same blackbox log. 

Any hints why this happens?

Greetings

mikeller, 2017-08-12:
What version is not working for you?

Grwmagad, 2017-08-12:
3.2.1

mikeller, 2017-08-12:
Can you try if it works for you with 3.2.0-RC3?

Grwmagad, 2017-08-12:
Ah you mean bf version, sorry.

I will try that.

Grwmagad, 2017-08-12:
Ok, 3.2 RC3 is working, at least with a really small log file. I will test a bigger one later.

Grwmagad, 2017-08-12:
Ok, 3.2 RC3 is working. 

Solved :-) 👍

closed 2017-08-12 by martinbudden