betaflight/betaflight-configurator #609: Restore migrates config to old version

jangrewe, 2017-08-12:
I'm using Configurator 3.2.1 and backed up my config because i needed to to a full chip erase.
After the erase, i wanted to restore my configuration - which did pretty much nothing.

I noticed this in the log:
```
2017-08-12 @ 19:49:29 -- Migrating configuration file generated by configurator: 3.2.1
2017-08-12 @ 19:49:29 -- Migrated configuration to configurator: 1.3.1
2017-08-12 @ 19:49:29 -- Configuration migration complete, migrations applied: 2
```

I'm not sure why Configurator 3.2.1 would want to migrate a config from Configurator 3.2.1 down to a config for Configurator 1.3.1?

Even if it's just a typo: it doesn't work! =(

mikeller, 2017-08-13:
@jangrewe: While the backup / restore in configurator is good for backing up and restoring (to the same firmware version, it can not be used to migrate the configuration during an upgrade.
For this purpose, please do a `diff all` in the CLI, save the output, and then copy / paste it into the CLI after the upgrade (copy the line with `defaults` separately).

@hydra: The message in the bug report looks like something in the migration code is picking up on versions that are after cleanflight 2.0. Can you please have a look?

jangrewe, 2017-08-13:
@mikeller I didn't do an upgrade, this was a backup from BF 3.2 RC3 restoring to BF 3.2 RC3

Like i said, i had to do a full chip erase to test something, but it was on the very same version.

hydra, 2017-08-13:
@mikeller I'm not in favour of using the current backup/restore functionality via MSP since it's not a complete backup.  Our time is better spent exposing PG's and backing/restoring up the config that way IMHO.  I don't currently have time to investigate, Maybe there's a simple fix though.

mikeller, 2017-08-13:
@hydra: I am having more of a 'tools for tasks' approach there: For a user who has only ever used the GUI to change settings, the GUI backup / restore will do a complete backup / restore of all non-default settings, and it's nice not to have to ask this user to use the CLI for backup / restore. The barrier to asking users who are using CLI anyway to adjust settings to use the CLI for backup / restore is much lower.

Anyhow, as a temporary fix to make GUI backup / restore usable again, how about making the whole config migration block conditional to the config being older than cleanflight 2.0? (What API version was that again?)

The alternative would be to remove the backup / restore buttons for these versions.

aschneider, 2017-08-30:
i am facing the exact same issue right now. And I can't even restore a backup I just made. There is simply no configuration at all applied. So if I for instance change PIDs in the Backup JSON and try to Restore from that JSON:

`2017-08-30 @ 14:27:59 -- Migrating configuration file generated by configurator: 3.2.1
2017-08-30 @ 14:27:59 -- Migrated configuration to configurator: 1.3.1
2017-08-30 @ 14:27:59 -- Configuration migration complete, migrations applied: 2`

Is there any solution to this? Or a workaround?

hydra, 2017-08-31:
@mikeller Sure go for the temporary workaround then, I looked for the API version list I made but couldn't find it.  It was on a configurator PR or issue though I forget if the comment was on BF or CF configurator.  You can check API versions by checking the cf files at each tagged release to find the api version.  It's not something I have committed to memory 😄

closed 2017-09-02 by mikeller

MagicHand1968, 2017-10-01:
I have the exact same issue as aschneider.

> 
i am facing the exact same issue right now. And I can't even restore a backup I just made. No Firmware Update, Same configurator Version. Simply export -> change some values e.g. PIDs -> restore -> nothing is applied.

2017-08-30 @ 14:27:59 -- Migrating configuration file generated by configurator: 3.2.1
2017-08-30 @ 14:27:59 -- Migrated configuration to configurator: 1.3.1
2017-08-30 @ 14:27:59 -- Configuration migration complete, migrations applied: 2
> 

Please reopen this Issue. I dont found any solution. Even the link #647 proposed by mikeller is not useable.

mikeller, 2017-10-05:
@MagicHand1968: See https://github.com/betaflight/betaflight-configurator/pull/647#issuecomment-334405427

PR #647 (2017-09-02): Fixed broken check for to migrate to version 1.3.1 (cleanflight).
Fixes #609.