betaflight/betaflight-configurator #615: What is the irc channel?

spuder, 2017-08-15:
The readme states that if you have questions you can reach out to the IRC channel. 

1. What is the IRC channel? 
2. Have you considered adding a betaflight configurator channel to the betaflight slack

I'd like to get started making a pull request to betaflight configurator but I'm not sure the best workflow for testing changes locally.

wind0r, 2017-08-15:
There isnt any IRC Channel. No one changed the info text, thats maybe a good first pull request from you to test stuff :) 

The Workflow is similar to https://github.com/betaflight/betaflight-configurator#alternative-way

Just get the Code, change it and install it as a unpacked extension to chrome.
Chrome will reload the folder on each code change. You just need to reload the App.

spuder, 2017-08-15:
I'll gladly change that. 

Shall I remove that line completely? 
Should we make a betaflight-configurator slack channel and point people there?

mikeller, 2017-08-16:
Not sure if we need a 'betaflight-configurator' slack channel. We had one in Gitter, before we moved to slack, and it never got much activity. So for now, just use 'general' or 'support' for discussion / questions about betaflight-configurator.

closed 2017-08-16 by spuder