betaflight/betaflight-configurator #617: Set Small Angle

spuder, 2017-08-16:
Since 3.2.x, `small_angle` is the only setting I have to set from the command line. 

Add support to set small angle via betaflight configurator gui

## Implementation details

- [ ] If accelerometer is disabled, grey out small_angle
- [ ] if accelerometer is enabled, activate small_angle
- [ ] Limit between 0 and 180
- [ ] Include Tip explaining what Small Angle does

## Discussion items

- [ ] In which section should this be located ( I think Configuration -> System Configuration tab)

- [ ] What order should this option be located (directly after `accelerometer`, or after `Magnetometer`)
- [ ] What to name the option (Should it be called `small_angle` or something more intuitive like `ARM Angle`)

---
Option 1: Located at bottom
[image]
Option 2: Located after accelerometer
[image]

mikeller, 2017-08-16:
Good suggestion!

As a name I would probably go for something like 'maximum arm angle'.

Also, please be aware that, since 3.2.0 is already in the RC phase and this will require a change in the firmware (to get / set the parameter value), this will be a change to go into 3.3.

spuder, 2017-08-16:
I actually think putting this under the ESC/Motor features makes more logical sense. 

[image]

martinbudden, 2017-08-16:
Why do you think it should be in the ESC/Motor features?

spuder, 2017-08-16:
The thinking behind the ECS/Motor features is that 2 of the options relate to arming. 

Perhaps it would be better to have a dedicated ARMing section
[image]

tigert, 2017-09-02:
I am pondering it should explain *what* the angle is in relation to (horizontal plane or whatever) - and then I realze I try to understand what is the function of small_angle? Why is there a defined max tilt angle that inhibits arming? For angle mode takeoff to avoid prop strikes?

spuder, 2017-09-08:
@tigert small_angle goes back to the days of multiwii https://github.com/multiwii/baseflight/wiki/CLI-Variables

For quad copters, I can't think of any reason you would want small_angle. Perhaps it was for safety when working on an aircraft? Perhaps to prevent prop strikes. 

I suggested removing small_angle entirely, or at least defaulting to 180 to the devs in the slack channel, but was met with some resistance. Making it easier to set is better than nothing

valerioangelici, 2017-09-22:
@spuder I think it could be a safety feature for beginners to have small_angle low at the beginning, as it prevents the copter from arming if you have it on your hand if you inadvertently flip the arming switch.
I agree that it should be configurable with the gui and not only from cli.

tigert, 2017-09-23:
@valerioangelici exactly, this is what I realized too - also the pretty common case of installing the flight controller flipped but failing to configure the board orientation accordingly - either because you are a new builder, or by mistake.

spuder, 2017-10-12:
Here is what I have working. 

[image]

closed 2017-10-14 by mikeller