betaflight/betaflight-configurator #619: Request

matoy, 2017-08-16:
Hi,

Would it be possible to implement BF3.2 new feature "crash recovery" to the configurator GUI?

See default related variables below:
set crash_dthreshold = 50
set crash_gthreshold = 400
set crash_setpoint_threshold = 350
set crash_time = 500
set crash_delay = 0
set crash_recovery_angle = 10
set crash_recovery_rate = 100
set crash_recovery = OFF

Thank you for all your work.

mikeller, 2017-08-17:
Thanks for changing the title to something more meaningful. This will go into 3.3. though, since 3.2 is already in the release candidate phase, which means that no new features will be added.

mosvov, 2017-10-16:
I can add that settings, 
as I understand it should be on PID's config tab?
also should we have new MSPCodes for that (MSP_CRASH_RECOVERY_CONFIG and MSP_SET_CRASH_RECOVERY_CONFIG) ?

mikeller, 2017-10-16:
@mosvov: Hold off on this. 3.3 will add support for MSPv2, and as part of that there will be support for reading / writing of parameter groups. This will remove the need to add custom MSP commands for config settings.

HThuren, 2023-07-17:
@haslinghuis I belive this are solved with return to home features ?

haslinghuis, 2023-07-17:
- Crash recovery as experimental feature is disabled by default and as such should not be exposed to UI. https://github.com/betaflight/betaflight/pull/2783

Some references:

- https://github.com/betaflight/betaflight/pull/7034
- https://github.com/betaflight/betaflight/pull/7924

closed 2023-07-17 by haslinghuis