betaflight/betaflight-configurator #625: Channel MAP letters are confusing

piepluquad, 2017-08-19:
Hello there, 
I always have to struggle when setting up Channel map in receiver tab converting the Aileron-Elevator-Rudder-Throttle to Roll-Pitch-Yaw-Throttle, just because i have to translate with these : 
Ailerons Control Roll axis, Elevator Controls Pitch axis, Rudder Controls Yaw axis.
Especially for not native english speaking people (i'm french by the way), this exercise is getting me tired...

May AETR letters switched to RPYT, so that figuring out what to change in configuration just by looking channel values when moving the sticks would be easier ?

some modifications i point : 

> tabs/receiver.js

 `var RC_MAP_Letters = ['A', 'E', 'R', 'T', '1', '2', '3', '4'];`
to : 
 `var RC_MAP_Letters = ['R', 'P', 'Y', 'T', '1', '2', '3', '4'];`

> tabs/receiver.html
```
                    <select class="hybrid_helper"
                        name="rcmap_helper">
                        <option value="AETR1234">Default</option>
                        <option value="AETR1234">FrSky / Futaba / Hitec</option>
                        <option value="TAER1234">Spektrum / Graupner / JR</option>
                    </select>

```
to : 
```
   <select class="hybrid_helper"
                        name="rcmap_helper">
                        <option value="RPTY1234">Default</option>
                        <option value="RPTY1234">FrSky / Futaba / Hitec</option>
                        <option value="TRPY1234">Spektrum / Graupner / JR</option>
                    </select>
```

then there wil need some kinda translation to AERT into js/msp/MSPHelper.js buffer handling, but i didn't get into there...

happy flying !

mikeller, 2017-08-19:
I don't think completely replacing AETR with RPTY is a good solution:
- AETR is what is used in other places, like the CLI, and the controller firmware;
- most users are probably used to, and more familiar with AETR.

We could either add a help text that shows the mapping between the two systems, or add a second, linked drop down for RPTY.

piepluquad, 2017-08-19:
well, actually if you put rudder input, you're going to yaw, so why input values (in bars 1000->2000) on the left are not Aileron Elevator Rudder ?
Roll Pitch Yaw are not actually inputs but quad behaviour...
just having two languages in hard IMHO

Arakon, 2017-08-20:
Also, AETR/TAER are used in many tutorials and manuals used for the transmitters and have been around for ages.

piepluquad, 2017-08-20:
OK, i don't want to change any habits, i just want to get life easier especially for newcomers : 
Can we imagine adding a hint letter to the bars in receiver tab, to help people figure out which channel to swap in 'Channel Map' entry ?

controlAxisRoll = Roll [A]
controlAxisPitch = Pitch [E]
controlAxisYaw = Yaw [R]
controlAxisThrottle = Throttle [T]
[image]

This is a minor modification to messages.json...

mikeller, 2017-08-20:
Yes, that's a really good idea! (Good UI often isn't about being very elaborate, but about getting it right.)

commit adb9aae5 (2017-08-21): Channel MAP letters are confusing (fixes #625)
closed 2017-08-21 by mikeller