betaflight/betaflight-configurator #630: Add sticks overlay to OSD

JonnyPhenomenon, 2017-08-22:
Feature request, to add the sticks to the OSD, kind of like the blackbox explorer can do, but in the FPV feed. 

Sure, the pilot really would have no need for this, as they would be watching where they were going, but this feature would be good for "passengers" monitoring your flight, especially in training scenarios, where an instructor would be able to watch the control inputs and coach the pilot, or while demonstrating a flight while showing the movements of the sticks... 

For example: (apologies for the crudeness of my sketch)

[image]

mikeller, 2017-08-25:
I assume you want this to happen in the live OSD generated by the quad, and not in the 'preview' shown in the configurator? If so, can you please close this, and re-open an issue in https://github.com/betaflight/betaflight/issues ? It sounds like a good idea, and we'll keep track of it in there.

JonnyPhenomenon, 2017-08-28:
Thanks Mike, I have opened the issue at https://github.com/betaflight/betaflight/issues/3988 and am closing this now.

closed 2017-08-28 by JonnyPhenomenon