betaflight/betaflight-configurator #632: Feature Request: Disallow or at least warn when disabling MSP on UART1 or VCP

joshuabardwell, 2017-08-22:
There is almost no scenario where one would want to disable MSP on VCP, or for a board with a CP210 chip, on UART1. I get so many support requests from people who have done this by accident, and the only fix is to re-flash the board and rebuild the config. For a beginner, this is a substantial hurdle.

Can we please have the configurator at least warn people that they will be locked out of their board if they disable MSP on VCP or UART1? Frankly, I could even see disallowing it entirely. Basically, prevent the user from locking themselves out of the board.

mikeller, 2017-08-22:
Hmm, I can see scenarios where users would want to move MSP to a different port. So as a counter-suggestion, how about detecting if **no** port is configured for MSP, and if so, enabling MSP on the first port (VCP or UART1)?

joshuabardwell, 2017-08-22:
To be completely honest, I think those scenarios are incredibly rare. But I do agree the user should have the option. So then I propose that there should be a pop-up warning if somebody is going to disable MSP on VCP or UART1. And they should be entirely prevented from locking themselves out. That seems like the best approach to me.

The warning might read:

WARNING! You are about to disable the USB port on your flight controller. If this is what you mean to do, press Ok. Otherwise, press Cancel.

I mean, make it totally clear.

joshuabardwell, 2017-08-22:
I don't think that re-enabling it on UART1 is a good approach. People won't expect the configurator to change their config like that. So for example if they have Serial RX on UART1, then they accidentally disable MSP on UART3 or something, now the receiver is getting MSP and potential issues could arise. 

I think the right approach is to simply prevent them from saving the config if they have disabled MSP entirely.

mikeller, 2017-08-23:
As I tried to explain in https://github.com/betaflight/betaflight-configurator/issues/632#issuecomment-324167873, there is no way for the user to completely disable MSP, since, if they remove it from all ports, the config validation will kick in, and reset the serial port config to defaults, resulting in MSP being reinstated on whatever port it is on by default.

piepluquad, 2017-08-24:
hi, here is my contribution : 
the pgResetFn_serialConfig in betaflight might be smarter ? for example i have a broken USB, and i manage to plug in my FTDI to access the board, shall it be possible to detect in betaflight that FC is powered up by an UART port with 5v, an so activate MSP on that UART ?

JonnyPhenomenon, 2017-08-24:
I have to side with Josh on this one.  You really need a big scary warning on this.  
The very first mistake I made on my very first quad was to accidentally disable the MSP on VCP. 
"OOOH, betaflight looks COOOOOOL... whats this VCP thing? pffff.  click... save.... WTF???"

I think it should have TWO stages of warnings, starting with Josh's 
WARNING! You are about to disable the USB port on your flight controller. If this is what you mean to do, press Ok. Otherwise, press Cancel.

And then one more that says "Oh boy!  you MIGHT want to save your config first... Since you wont be able to talk to your flight controller again without completely reflashing it first.  And if you really want to brick your flight controller, you might want to write this web address down.  https://oscarliang.com/cannot-connect-to-naze32-brick-fix/ "
with these two buttons "HURT ME" and "Oh wait, what do you mean BRICK??"

add a 30 second timer with an increasingly terrifying count down alarm too.

JonnyPhenomenon, 2017-08-24:
@piepluquad  I dont think the FC has any way to detect it is being powered by the 5v from a specific uart. all of the 5v pins are the same, just like the ground plane.

piepluquad, 2017-08-24:
@JonnyPhenomenon you're right... can't betaflight detect MSP traffic incoming from an RX pin/pad ?

mikeller, 2017-08-24:
@JonnyPhenomenon: That must have been an older version of Betaflight then. In current Betaflight, if you change the settings so that there is not at least one port with MSP enabled, this will be detected and MSP will be re-enabled on the default port (VCP for boards with VCP, UART1 for the rest).

@piepluquad: The problem with 'detecting' MSP is that there is very little structure in MSP messages, so differentiating between MSP and a stream of data that is not MSP (say, SBUS from your RX) is hard. We've already had to disable MSP listening on UART1 by default for boards with VCP, since this caused lockups on boot for users with a serial RX on UART1, because the RX data stream was misinterpreted as MSP.

piepluquad, 2017-08-24:
Ok, so this may be tagged not reproducible and thread closed ?

JonnyPhenomenon, 2017-08-28:
@mikeller : Indeed Mike, it was about a year ago, and might even have been with cleanflight.  
You mentioned earlier that the the MSP port gets reset to default if it got _disabled_ by accident.  But if someone changed it to a different port by accident, wouldn't a warning at least be helpful? 

Even if it wasn't a confirmation dialog, but some visible text on the configuration tab

mikeller, 2017-08-29:
@JonnyPhenomenon: I doubt that a visible warning will help. As it looks, nobody is reading them:

[image]

JonnyPhenomenon, 2017-08-29:
@mikeller HAH!
well isn't that something. staring right at us the whole time... :)

closed 2018-05-13 by stale[bot]