betaflight/betaflight-configurator #636: Feature:ADD Android Support

Stavr0s91i, 2017-08-28:
Isnt it time someone ported this over to android or ios? It should be that hard using Apache Cordova is someone has some experience with it

tigert, 2017-09-02:
This is an open source project created by people like you mostly on their spare time. Maybe you can give it a try?

Woody9999, 2017-12-08:
With release of the new Spektrum transmitter that is capable of running Android apps, the ability to run the configurator app would be very beneficial, with a potentially huge user base.

I'm just a user, and have no experience to be able to  program such a project.

cmengler, 2018-05-08:
I've made progress with Apache Cordova using the latest Configurator v10.3.0 source and USB OTG for connectivity -- there is also the possibility for iOS using WiFi serial modules later on.

Just wanted to open up further discussions on some approaches regarding the UI layout.

My thoughts is to start with making the Configurators layouts be responsive for smaller screen sizes, use something lightweight like Pure CSS grid layout system.

McGiverGim, 2018-05-08:
I don't know if you have seen the speedybee app https://appadvice.com/app/speedybee-app/1150315028

I'm pretty sure they are breaking the GPL, but what they have done is what you are trying to do. Maybe you can take ideas from there.

mikeller, 2018-05-08:
> SpeedyBee is an Appllication for Flight controller parameters settings

They are also breaking the rules of English orthography...

haslinghuis, 2021-06-05:
We have Android releases based on Cordova now. But support is brittle as no one seems interesting in supporting this.

closed 2021-06-05 by haslinghuis