betaflight/betaflight-configurator #646: Feature idea: "Channel Bug" logo for OSD

tigert, 2017-08-31:
Wrote this down after some discussion on slack today.

We have currently some free space in the OSD chip character memory that is used to display the Betaflight logo on startup. It seems to be 24 characters in 3 rows. What if we used part of this as "channel bug" feature? Basically an abitary small bitmap you could put on your OSD to personalize your DVR footage, for example.

You can currently hack this using the "aircraft name" setting by naming your craft for example "XYZ" and then make a customized font where you draw your logo over the XYZ characters. And hope that nothing else in the OSD uses those letters :-)

Ideally you could select a PNG image with the configurator with some maximum size (48x54 for example, so it uses 4x3 character grid) and upload it as custom characters to the OSD chip. 

You could then place your "pilot art" as a data field on the OSD wherever you wanted.

mikeller, 2017-08-31:
I like the idea. I think before we address this, we should get the font optimisations that are proposed in betaflight/betaflight#3461 and #541 implemented (including a 'header character' containing metadata about the font). This will give us enough space in the first bank to implement the custom logo for all OSD chips.

tigert, 2017-09-01:
A bit of inspiration and a default bug (pun intended :))

[image]

[image]

It might be a bit large, but chopping to 2x2 characters might be a bit too small as well, so I guess the designs would need to be placed towards the topright corner in case you want a slightly smaller logo (unless we want to implement exact positioning in the OSD settings, I guess that would be possible too, if you map it to the underlying character grid and then endode it into a font on the fly when you upload it?

tigert, 2017-09-01:
A bit smaller version - though we start to lose clarity fast if we shrink it further.

[image]

wind0r, 2017-09-01:
This PR: https://github.com/betaflight/betaflight/pull/3461 also will allow way more free spaces for logos and stuff :)

tigert, 2017-09-01:
In case you want to replace the current logo in the fonts with a bit more pixel-aligned version that has the contrast outline too, so it looks better over the video.

[image]

tigert, 2017-09-01:
Sorry @jimmac I just had to ;-)

[image]

mikeller, 2017-09-03:
@tigert: We need transitions based on <ground speed> / <altitude> according to this scale: https://i.imgur.com/9w0SPNn.jpg

tigert, 2017-09-04:
First I was like, waiting for the race to start..
[image]

Noooo!
[image]

Would be fun and awesome, but I doubt the readability would be useful.

However, animating your avatar from series of images based on a variable (like bank angle or throttle or aux channel ;-) ) would be pretty awesome.

closed 2018-05-13 by stale[bot]