betaflight/betaflight-configurator #649: "Camera control" Modes displaying old/wrong text first time

Arakon, 2017-09-02:
Runcam Split configured on UART2.
Connecting to current 3.2.0 (as of today) using the latest (as of today) configurator and clicking on modes results in this screen:
[image]
If I then click on another tab (i.e. receiver or configuration) and back on the modes tab, the names change to the correct ones:
[image]

tigert, 2017-09-05:
Dumb question, but you clicked "Reload" in your git-cloned configurator in the Extensions config on Chrome after you did a git pull?

Arakon, 2017-09-05:
Yup. No difference. Issue was confirmed in the runcam split thread on RCG by someone else too.

basdelfos, 2017-09-05:
The box names for the runcam split commands are translated in code.
https://github.com/betaflight/betaflight-configurator/blob/13e505ec61d99fe227d91c77a579264bbddc2089/tabs/auxiliary.js#L39-L50

Somehow there is a condition when the runcam split box names are only translated when runcam split is selected as peripheral in the ports tab. This is wrong, the text should always be translated even when runcam split is not selected as peripheral.

https://github.com/betaflight/betaflight-configurator/blob/13e505ec61d99fe227d91c77a579264bbddc2089/tabs/auxiliary.js#L58-L60

I will submit a pr to fix this and also move the hardcoded translation to the localized resource file.

PR #657 (2017-09-06): Fix Runcam Split command box names and move to localized resource file
Fixes #649
commit ca12f53e (2017-09-06): Fix Runcam split command box names and move to localized resource file. Fixes #649.
closed 2017-09-06 by mikeller