betaflight/betaflight-configurator #652: OSD screen layout for buttons is broken

Arakon, 2017-09-03:
Configurator 3.2.2 (current master) is missing the save and refresh (?) buttons in the bottom right.
[image]

AndersHoglund, 2017-09-03:
My copy of BFC 3.2.2 (current master HEAD = 7104a95edcef23a1240d78206dc696339127c489) works just fine.  Checked on two Win10 machines, one small low-res Laptop screen and one large hi-res stationary. Both in windowed and fullscreen modes.

Arakon, 2017-09-04:
I just re-downloaded the current master and overwrote the folder again and it still happens. Switch to official 3.2.1 release and the buttons are there (Font Manager and Save, not Refresh as I stated above).

closed 2018-05-13 by stale[bot]