betaflight/betaflight-configurator #658: V3.2.1 - Newly added copy Profiles and Rateprofiles do not work

chime13, 2017-09-06:
Using Naze 3.20 RC4 on Scisky v1.2FC. All manually inputed data in GUI or CLI work as intended. When using the new copy buttons though, if I copy profile 1 to profile 2, actual copy does not take place. I tried to save the setting after copying the profile or rateprofile, but the old values are still present.

Arakon, 2017-09-06:
I believe the function on the firmware side was not part of RC4 yet. The pull request for that wasn't merged until august 31st.

chime13, 2017-09-15:
Updated 5 FCs to 3.20RC5 on BC 3.2.2 and still the functions are flaky. Also, unless I'm wrong, updating rateprofile thru the GUI results in all rateprofiles of the same number getting updated for all profiles. For example: I select [Profile 1, Rateprofile 1] - if input a new rate profile setting, [Profile 2, Rateprofile 1] and [Profile 3, Rateprofile 1] also get the new setting. There is no way of having individual rateprofiles per profile.

mikeller, 2017-09-15:
@chime13: With 3.2, there are 3 profiles (2 on F1), and 3 rateprofiles, and they are independent. (i.e. the rateprofiles are no longer stored per profile)

chime13, 2017-09-15:
@mikeller Thanks for the update - now I'm clear at least on that aspect. One thing though, I do have 2 F1 Sciskys that now store 3 profiles. I verified it using stick commands - I can see the LED blink according to the profile selected - too cool!

mikeller, 2017-09-15:
Not sure if that actually works (assuming you are referring to profiles and not rateprofiles) - the intention certainly was to support 2 of them on F1. So please make sure profile 3 is not just reverting to either 1 or 2 and it's a bug.

chime13, 2017-09-15:
Here's a "Diff All" from one of my Sciskys. Again, verified 3 different profile selections - with both, GUI and LED on FC.

2017-09-15 @ 09:57:41 -- Running - OS: MacOS, Chrome: 60.0.3112.113, Configurator: 3.2.2
2017-09-15 @ 09:57:47 -- Serial port successfully opened with ID: 1
2017-09-15 @ 09:57:47 -- MultiWii API version: 1.36.0
2017-09-15 @ 09:57:47 -- Flight controller info, identifier: BTFL, version: 3.2.0
2017-09-15 @ 09:57:47 -- Running firmware released on: Sep 11 2017 06:42:59
2017-09-15 @ 09:57:47 -- Board: AFNA, version: 2
2017-09-15 @ 09:57:47 -- Unique device ID: 0x674ff535353667267125633
2017-09-15 @ 09:57:47 -- Craft name: Smart 100
2017-09-15 @ 09:57:55 -- Loaded Profile: 2
2017-09-15 @ 09:57:57 -- Loaded Profile: 3
2017-09-15 @ 09:58:00 -- Loaded Profile: 2
2017-09-15 @ 09:58:05 -- Loaded Profile: 1
2017-09-15 @ 09:58:08 -- Loaded Profile: 2
2017-09-15 @ 09:58:15 -- EEPROM saved
2017-09-15 @ 09:58:17 -- Loaded Profile: 3
2017-09-15 @ 09:58:21 -- CLI mode detected
Setup
Ports
Configuration
Power & Battery
Failsafe
PID Tuning
Receiver
Modes
Adjustments
Servos
Motors
Sensors
Tethered Logging
Blackbox
CLI
Note: Leaving CLI tab or pressing Disconnect will automatically send "exit" to the board. With the latest firmware this will make the controller restart and unsaved changes will be lost.

CLI

# version
# Betaflight / NAZE (AFNA) 3.2.0 Sep 11 2017 / 06:42:59 (649ccd7c7) MSP API: 1.36

# diff all
# Betaflight / NAZE (AFNA) 3.2.0 Sep 11 2017 / 06:42:59 (649ccd7c7) MSP API: 1.36
defaults
name Smart 100

feature MOTOR_STOP
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -MULTI_BEEPS
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
aux 0 28 1 1300 1700
adjrange 0 0 0 1700 2075 12 0
rxrange 0 1067 1964
rxrange 1 1067 1964
rxrange 2 1067 1964
rxrange 3 1067 1964
set gyro_sync_denom = 3
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set acc_hardware = NONE
set baro_hardware = NONE
set min_check = 1013
set max_check = 1950
set fpv_mix_degrees = 20
set airmode_start_throttle = 1250
set min_throttle = 1013
set use_unsynced_pwm = ON
set motor_pwm_protocol = BRUSHED
set motor_pwm_rate = 4000
set yaw_motors_reversed = ON
set auto_disarm_delay = 1
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 2
profile 0

set anti_gravity_gain = 3000
set setpoint_relax_ratio = 40
set dterm_setpoint_weight = 70
set p_pitch = 130
set i_pitch = 60
set d_pitch = 130
set p_roll = 130
set i_roll = 60
set d_roll = 130
set p_yaw = 130
set i_yaw = 100
set d_yaw = 30
profile 1

set anti_gravity_gain = 3000
set setpoint_relax_ratio = 40
set dterm_setpoint_weight = 70
set p_pitch = 130
set i_pitch = 60
set d_pitch = 140
set p_roll = 130
set i_roll = 60
set d_roll = 140
set p_yaw = 130
set i_yaw = 100
set d_yaw = 30
profile 2

set anti_gravity_gain = 3000
set setpoint_relax_ratio = 40
set dterm_setpoint_weight = 70
set p_pitch = 135
set i_pitch = 60
set d_pitch = 140
set p_roll = 135
set i_roll = 60
set d_roll = 140
set p_yaw = 130
set i_yaw = 100
profile 2
rateprofile 0

set rc_rate = 124
set rc_rate_yaw = 128
set rc_expo = 50
set thr_mid = 40
set pitch_srate = 73
set yaw_srate = 73
rateprofile 1

set rc_rate = 214
set rc_rate_yaw = 213
set rc_expo = 60
set thr_mid = 40
set thr_expo = 20
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 0
rateprofile 2

set rc_rate = 124
set rc_rate_yaw = 128
set thr_mid = 40
set pitch_srate = 73
set yaw_srate = 73
rateprofile 0
save
#

chime13, 2017-09-15:
Just wanted to update:
Seems the copy functions are indeed working correctly. I was unaware of the fact that the RateProfiles were no longer tied to individual Profiles. Now RateProfiles are globally adjusted regardless of the profile. Personally, I like having my Rates tied to my Profiles, but at least it's working perfectly.

Again @mikeller, I have 2 Scisky v1.2 F1 DSM FCs that prior to this release, had only 2 selectable/tunable Profiles. We now have 3 Profiles available for this board. Thanks Again!

closed 2017-09-15 by chime13