betaflight/betaflight-configurator #661: 32khz gyro sampling causes incorrect blackbox log rate options

etracer65, 2017-09-10:
If 32khz gyro sampling is enabled, then the blackbox logging rates are incorrect for on-board flash.  Consider the following configuration:

[image]

Then the only options in the blackbox tab are:

[image]

If a rate is selected in the cli for a faster speed like 8khz (`blackbox_p_denom = 256`) then the blackbox tab shows an empty selection for the rate:

[image]

The rate options in the blackbox tab appear to behave normally when 32khz sampling is not selected.

mikeller, 2017-09-13:
Thanks, @basdelfos.

basdelfos, 2017-09-14:
I can reproduce this issue also without 32khz enabled (as I don't own a 32khz capable fc). I tried it on an OMNIBUSF4 by settings `set blackbox_p_denom = 256` in the cli. The blackbox tab shows the same empty dropdown box.

martinbudden, 2017-09-14:
Yes, `blackbox_p_denom` should be checked in `validateAndFixConfig`.

mikeller, 2017-09-15:
> The fw shouldn't allow to set the blackbox_p_denom setting to a higher rate than the active pidloop speed, or am I wrong?

To my understanding, this is an intrinsic property of `blackbox_p_denom` being a denominator, and being based off the PID loop. (i.e. `blackbox_p_denom` of `1` means a P frame gets logged for every PID loop iteration, `2` means every second PID loop iteration, and so forth)

martinbudden, 2017-09-15:
@mikeller , the way the denominator works has changed in 3.2.

Blackbox has two main types of frames, I-frames which store the absolute data, and P-frames which store deltas (so P-frames are smaller). In 3.2 blackbox was changed so that there are 32 I-frames per second, regardless of PID loop speed. The blackbox logging rate is then determined by the number of P-frames per I-frame (`blackbox_p_denom`). So if the p-denom is 16, then a blackbox frame will be logged  32*16= 512 times a second.

mikeller, 2017-09-15:
@martinbudden: Can I just say that this is a convoluted way to set frame rates at the moment:

https://github.com/betaflight/betaflight/blob/07d599039550fc35738d0e7a7e740d6badad5de5/src/main/blackbox/blackbox.c#L1703-L1708

Also, n.b. that `blackboxConfig()->p_denom` is now a denominator on period, as opposed to `gyroConfigMutable()->gyro_sync_denom` and `pidConfigMutable()->pid_process_denom` that are still denominators on frequency. I think it should be renamed to make it apparent that they now work in opposite directions.

basdelfos, 2017-10-09:
In 3.2.0 official release the blackbox_p_denom value is not there anymore (at least in cli). Is the BFC also changes for that or is that not necessary?

McGiverGim, 2019-04-03:
@etracer65 I think this is and old issue. I suppose it is fixed and we can close it?

etracer65, 2019-04-03:
Not sure if it's resolved but since we don't support 32KHz sampling anymore I guess it's no longer an issue. Closing.

closed 2019-04-03 by etracer65