betaflight/betaflight-configurator #668: Restore button does not work in 3.2RC5

slacker87, 2017-09-15:
The backup file gets properly created, but when restore is attempted it does not restore the previous config state. Restore will prompt for the restore file, accept it, log `Configuration migration complete, migrations applied: 2`, but the config is never actually restored.

MR444, 2017-10-11:
I have the same problem after flashing from 3.1.5 to 3.2.0. Size of json files is always 33KB. Restoring backup files does not change anything.

mikeller, 2017-10-11:
Fixed by #647 (still needs to be released to the Chrome web store).

dthubereng, 2017-12-06:
I noticed the same thing as the OP on 3.2.2.

Restore works mostly in 10.0.0 for me with the exception that it doesn't restore everything. So far, I've noticed it doesn't restore ESC/motor protocol and yaw position (haven't looked at the other positions because yaw is the only one I have to adjust on my build). I've checked this with both Mac OS X High Sierra and Win 10.

mikeller, 2017-12-06:
@dthubereng: The way the save / restore buttons work is not done in a great way, since it relies on all of the data explicitly being polled from the flight controller before serialising and saving, and then being pushed after loading. I'll look into adding the missing MSP calls for this (or you can if you want to).
In the mean time, use the CLI for back ups / restores.

dthubereng, 2017-12-06:
not real familiar with js, but I've been looking at the backup_restore functions to see where the missing calls are.

mikeller, 2017-12-06:
@dthubereng: No problem then, I can do it. `backup_restore.js` isn't exactly the best place to be doing a deep dive into JS.

closed 2018-05-13 by stale[bot]