betaflight/betaflight-configurator #682: Failure to Configure after Flash Firmware on two platforms

GladiatorDev, 2017-10-10:
I believe this is a software issue with BetaFlight 3.2.1. Before this version we never experienced this problem. We have two completely different quads. After flashing the latest stable firmware on both, BetaFlight 3.2.1 will tells us that it successfully connects to the quad then immediately it says that it successfully closes the serial port - disconnecting it. The exact same problem was reproducible on both Windows 10 and the latest MacOS. Is this a bug in 3.2.1?
Thanks!

jflyper, 2017-10-19:
Please follow the guideline for issue submission.

jflyper, 2017-10-19:
What is your FC?

GladiatorDev, 2017-10-19:
We determined it was a hardware problem. Next time I'll make sure to follow guidelines when I post something. Thanks!

closed 2017-10-24 by jflyper