betaflight/betaflight-configurator #690: Exiting CLI doesn't reboot cleanly

richard-scott, 2017-10-15:
When exiting the CLI, the GUI crashes. 
It used to disconnect and reconnect nicely, but now it doesn't.

I'm using the codebase from commig  a976b88 on the 15th October.

[image]

richard-scott, 2017-10-15:
It's the same when saving blackbox settings too:
[image]

mikeller, 2017-10-15:
Try pulling the latest configurator - there was a bug in the disconnection code, it was fixed in #688 / #689.

richard-scott, 2017-10-15:
My Laptop pulls down the latest codebase from master every time it boots. I've just booted, and tried now and still get the same. All I did this time was connect to the FC (an SP Racing F3 Evo), click Ports, and click Save and Reboot and got this in the screen shot.

[image]

mikeller, 2017-10-15:
Just tested, works fine for me. What operating system is running on your Laptop? Can you manually pull latest for the configurator to make sure your update is working properly?

richard-scott, 2017-10-17:
I've tried it again today after pulling the latest code and it looks to be ok now.

closed 2017-10-17 by richard-scott