betaflight/betaflight-configurator #692: Can't set PID loop frequency, Board: CLR4

srw2d, 2017-10-19:
Hi,

Just recently got a clracing f4. Installed BF 3.2.1 but also tried 3.2. It won't let me set the PID loop frequency beyond 2 kHz. I was able to in CLI to "set pid_process_denom = 1" and that worked. The configurator works for everything else so far in the configuration tab. The CPU load is at 4% when I run 8k/8k and I was interested in trying to enable gyro 32khz sampling mode but not sure that is an option for this board. The bigger issue is that I can't adjust the PID loop frequency in the configurator. Not sure if its a hardware issue or a bug. Going to have a friend check with his board as well, but thought I should report it.

I tried on both Linux and Windows OS. 
Configurator version is 3.2.1

Thanks!

srw2d, 2017-10-19:
Here is a dump 

# version
# Betaflight / CLRACINGF4 (CLR4) 3.2.1 Oct 15 2017 / 19:29:56 (434db9a36) MSP API: 1.36

# name
name -

# resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 B08
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 B08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource INVERTER 1 C00
resource INVERTER 2 NONE
resource INVERTER 3 NONE
resource INVERTER 4 NONE
resource INVERTER 5 NONE
resource INVERTER 6 NONE
resource INVERTER 7 NONE
resource INVERTER 8 NONE
resource INVERTER 9 NONE
resource INVERTER 10 NONE
resource INVERTER 11 NONE
resource LED 1 B05
[... 542 lines cut ...]
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 65
set rc_expo_yaw = 55
set thr_mid = 50
set thr_expo = 0
set roll_srate = 83
set pitch_srate = 83
set yaw_srate = 82
set tpa_rate = 10
set tpa_breakpoint = 1650

#

srw2d, 2017-10-19:
I was able to fix this by changing:

`set acc_hardware = AUTO`

to 
`set acc_hardware = NONE`

After I did this, settings take in the configurator, even turning accelerometer on and off the settings still stay. I'm confused as to why this would be but its working now.

mikeller, 2017-10-19:
It could have been caused by a corrupted configuration. If you get 'stuck' settings, doing a 'Reset to defaults', and then restoring the settings usually helps.

If this is fixed now, can you please close the issue?

srw2d, 2017-10-19:
@mikeller sure thanks, will close.

closed 2017-10-19 by srw2d

byteconcepts, 2017-11-05:
Hi,

sorry, but it seems this issue has to be reopened...

I just got a FrSky XSRF4O FC/XSR AIO board which was shipped with an OMNIBUSF4 3.1.7 target.
The "instruction manual" says, when flashing the FC firmware, the FRSKYF4 (FRF4) target should be chosen.

I then managed to flash it also to the new BF 3.2.1 FRSKYF4 target but get the same issue.

Unfortunately, even resetting it to the defaults or setting acc_hardware = NONE or to MPU6000 did not enable saving pid_process_denom = 1.
I tried all that in different combinations, even re-flashing it, setting it to the defaults afterwards and then turning accelerometer and baro off. - It just won't save a higher loop frequency then 2khz, with gui or cli.

Any more ideas how to solve this problem? - It's an F4 board so it should absolutely be possible to run 8khz loop frequency.