betaflight/betaflight-configurator #712: Can't set PID loop frequency, Board: FRSKYF4 (FRF4)

byteconcepts, 2017-11-05:
Hi,

I just got a FrSky XSRF4O FC/XSR AIO board which was shipped with an OMNIBUSF4 3.1.7 target.
The "instruction manual" says, when flashing the FC firmware, the FRSKYF4 (FRF4) target should be chosen.

I then managed to flash it also to the new BF 3.2.1 FRSKYF4 target but get the same issue.

Unfortunately, even resetting it to the defaults or setting acc_hardware = NONE or to MPU6000 did not enable saving pid_process_denom = 1.
I tried all that in different combinations, even re-flashing it, setting it to the defaults afterwards and then turning accelerometer and baro off. - It just won't save a higher loop frequency then 2khz, with gui or cli.

Any more ideas how to solve this problem? - It's an F4 board so it should absolutely be possible to run 8khz loop frequency.

wind0r, 2017-11-05:
Motor Protocol Oneshot125 will not allow you any higher looptime then 2khz

byteconcepts, 2017-11-05:
Sorry for that,

I switched to dshot 600 now after flashing the XSR firmware to LBT and now it works.

closed 2017-11-05 by byteconcepts