betaflight/betaflight-configurator #739: 10.0.0 RC1 MacOS

jcutcher01, 2017-11-24:
Trying to run the 10.0.0 RC1 stand-alone configurator on Mac and its crashing. I right-click open and then with the warning message click open and it says stopped unexpectedly.

Process:               nwjs [769]
Path:                  /private/var/folders/*/Betaflight Configurator.app/Contents/MacOS/nwjs
Identifier:            com.nw-builder.betaflight-configurator
Version:               Version 10.0.0-RC1 (10.0.0-RC1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           nwjs [769]
User ID:               501

Date/Time:             2017-11-24 13:49:35.341 -0500
OS Version:            Mac OS X 10.13.1 (17B48)
Report Version:        12
Anonymous UUID:        D4D6E988-3BBF-B824-AC17-A07B70ADE600

Time Awake Since Boot: 710 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff58d0ffce __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff58e4d150 pthread_kill + 333
2   libsystem_c.dylib             	0x00007fff58c6c30a abort + 127
3   com.nw-builder.betaflight-configurator	0x0000000102214d95 main + 325
4   libdyld.dylib                 	0x00007fff58bc0145 start + 1

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x00007fff92213340  rcx: 0x00007ffeed9eba88  rdx: 0x0000000000000000
  rdi: 0x0000000000000307  rsi: 0x0000000000000006  rbp: 0x00007ffeed9ebac0  rsp: 0x00007ffeed9eba88
   r8: 0x00007fff921f3048   r9: 0x0000000000000040  r10: 0x0000000000000000  r11: 0x0000000000000206
  r12: 0x0000000000000307  r13: 0x00006040001bb200  r14: 0x0000000000000006  r15: 0x000000000000002d
  rip: 0x00007fff58d0ffce  rfl: 0x0000000000000206  cr2: 0x00007fff921f1148
  
Logical CPU:     0
Error Code:      0x02000148
Trap Number:     133

Binary Images:
       0x102214000 -        0x102214ffb +com.nw-builder.betaflight-configurator (Version 10.0.0-RC1 - 10.0.0-RC1) <4C1576E6-8806-3045-B908-23FBC0980646> /var/folders/*/Betaflight Configurator.app/Contents/MacOS/nwjs
       0x10efc6000 -        0x10f01098f  dyld (519.2.1) <CD85EFBA-B9D6-3FD5-B777-A23B2DDEF192> /usr/lib/dyld
    0x7fff2db81000 -     0x7fff2db81fff  com.apple.Accelerate (1.11 - Accelerate 1.11) <5AA750F5-D633-32BA-B7F3-4F651FB1761E> /System/Library/Frameworks/Accelerate.framework/Versions/A/Accelerate
    0x7fff2db99000 -     0x7fff2e097fc3  com.apple.vImage (8.1 - ???) <310976EE-E12D-39D7-8F58-6EE924E08576> /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vImage.framework/Versions/A/vImage
    0x7fff2e098000 -     0x7fff2e1f2fcb  libBLAS.dylib (1211.20.1) <B70FC2EE-0859-39DB-927E-0020AE96907F> /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libBLAS.dylib
    0x7fff2e1f3000 -     0x7fff2e220fef  libBNNS.dylib (32) <9CA15DC6-004A-32FD-BFCA-F5D488012C43> /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libBNNS.dylib
    0x7fff2e221000 -     0x7fff2e5e1ff7  libLAPACK.dylib (1211.20.1) <04B937A4-D0E7-33C8-A1C3-2724AD4FAB2F> /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libLAPACK.dylib
    0x7fff2e5e2000 -     0x7fff2e5f7ff7  libLinearAlgebra.dylib (1211.20.1) <355307DE-82A5-32A3-8AC6-603D90E99601> /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libLinearAlgebra.dylib
    0x7fff2e5f8000 -     0x7fff2e5fdff3  libQuadrature.dylib (3) <3D6BF66A-55B2-3692-BAC7-DEB0C676ED29> /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libQuadrature.dylib
    0x7fff2e5fe000 -     0x7fff2e65cfff  libSparse.dylib (79.1.1) <7AD0F8A8-FD36-36FE-B83D-58648EBD0027> /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libSparse.dylib
    0x7fff2e65d000 -     0x7fff2e670fff  libSparseBLAS.dylib (1211.20.1) <9C7AB67B-5BFD-3571-8D90-312D1DB10098> /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libSparseBLAS.dylib
    0x7fff2e671000 -     0x7fff2e81dfc3  libvDSP.dylib (622.20.8) <6FFCA52B-7D60-326A-ADF2-601F39A8685A> /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libvDSP.dylib
    0x7fff2e81e000 -     0x7fff2e8cefef  libvMisc.dylib (622.20.8) <54F90047-879F-3260-8604-6E453149B49E> /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libvMisc.dylib
    0x7fff2e8cf000 -     0x7fff2e8cffff  com.apple.Accelerate.vecLib (3.11 - vecLib 3.11) <8A96A8ED-7B88-3D17-8D17-41D224E0EC90> /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/vecLib
    0x7fff2ebb9000 -     0x7fff2fa14ff7  com.apple.AppKit (6.9 - 1561.10.101) <AC5C259A-615C-3929-8586-0BC920696AFA> /System/Library/Frameworks/AppKit.framework/Versions/C/AppKit
    0x7fff2fa66000 -     0x7fff2fa66fff  com.apple.ApplicationServices (48 - 50) <3EB38526-9975-36C9-BADC-F9019A608267> /System/Library/Frameworks/ApplicationServices.framework/Versions/A/ApplicationServices
    0x7fff2fa67000 -     0x7fff2facdfff  com.apple.ApplicationServices.ATS (377 - 445) <2303339C-5E2A-37DB-B2B0-E4E5DA9121E5> /System/Library/Frameworks/ApplicationServices.framework/Versions/A/Frameworks/ATS.framework/Versions/A/ATS
    0x7fff2fb66000 -     0x7fff2fc88ff7  libFontParser.dylib (222.1.1) <56C57A24-6A6A-3F32-8E40-5BFA4DA4B04D> /System/Library/Frameworks/ApplicationServices.framework/Versions/A/Frameworks/ATS.framework/Versions/A/Resources/libFontParser.dylib
    0x7fff2fc89000 -     0x7fff2fcd3ff7  libFontRegistry.dylib (221) <A22F82C0-B4FE-3DB5-B968-79B28257DF2F> /System/Library/Frameworks/ApplicationServices.framework/Versions/A/Frameworks/ATS.framework/Versions/A/Resources/libFontRegistry.dylib
    0x7fff2fe15000 -     0x7fff2fe19ff3  com.apple.ColorSyncLegacy (4.13.0 - 1) <7C8AD42D-5A89-3BA6-B55D-374D9267E939> /System/Library/Frameworks/ApplicationServices.framework/Versions/A/Frameworks/ColorSyncLegacy.framework/Versions/A/ColorSyncLegacy
[... 218 lines cut ...]
__DATA                            20.7M      207 
__FONT_DATA                          4K        2 
__LINKEDIT                       187.2M        4 
__TEXT                           129.9M      212 
__UNICODE                          560K        2 
shared memory                        8K        3 
===========                     =======  ======= 
TOTAL                            425.6M      450 

Model: MacBookPro11,1, BootROM MBP111.0142.B00, 2 processors, Intel Core i5, 2.6 GHz, 8 GB, SMC 2.16f68
Graphics: Intel Iris, Intel Iris, Built-In
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, 0x802C, 0x384B54463531323634485A2D314736453120
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x802C, 0x384B54463531323634485A2D314736453120
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x112), Broadcom BCM43xx 1.0 (7.77.37.2.1a3)
Bluetooth: Version 6.0.0f1, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
PCI Card: pci11c1,5901, IEEE 1394 Open HCI, Thunderbolt@195,0,0
Serial ATA Device: APPLE SSD SM0512F, 500.28 GB
USB Device: USB 3.0 Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
FireWire Device: Vendor 0x1F2 Device 0x107800, 0x1F2, Up to 400 Mb/sec
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.2
Thunderbolt Device: Thunderbolt to FireWire Adapter, Apple Inc., 1, 5.6

mikeller, 2017-11-24:
@jcutcher01: Thank you very much for the report. This problem has been identified, and we'll release 10.0.0-RC2 shortly, which will fix the osx64 platform.

jcutcher01, 2017-11-25:
Excellent.

AliB10, 2017-11-26:
I've just tried to load RC2 OSX and mine just hangs on verification window.

Tried putting build into applications folder, and running straight from the drive and it does the same.
RUnning El Capitan if that makes any difference.

jcutcher01, 2017-11-26:
I am running a 2013 MacBook Pro with the latest OS and updates and RC2 installed and runs.

AliB10, 2017-11-26:
ahh. left it for 15 minutes and it finally opened. Now opens quickly :)

closed 2017-11-26 by mikeller