betaflight/betaflight-configurator #742: Blackbox download doesn't seem to work

SandroGrzicic, 2017-11-25:
On BF Configurator master (10.0.0-RC1), trying to download Blackbox from my fishpepper TinyFC, the Saving dataflash to file window stays open forever with the progress at 0%, and the console repeating the following messages.

The tinyfish is running Betaflight 3.2.2 with target TFSH.

At first I thought that the dataflash chip is bad, but I replaced it with another one and I get identical behaviour.

````
Dataflash dump file path: E:\Quadcopter\TomoquadsKosho\Blackbox\BTFL_BLACKBOX_LOG_20171125_001127.BBL
msp.js:210 MSP data request timed-out: 71
msp.js:210 MSP data request timed-out: 116
msp.js:210 MSP data request timed-out: 150
msp.js:210 MSP data request timed-out: 110
msp.js:210 MSP data request timed-out: 71
msp.js:210 MSP data request timed-out: 150
msp.js:210 MSP data request timed-out: 116
msp.js:210 MSP data request timed-out: 110
msp.js:117 code: 71 - crc failed
MSPHelper.js:1610 CRC error for address 0 - retrying
msp.js:210 MSP data request timed-out: 71
msp.js:210 MSP data request timed-out: 116
msp.js:210 MSP data request timed-out: 150
msp.js:210 MSP data request timed-out: 110
msp.js:210 MSP data request timed-out: 71
msp.js:210 MSP data request timed-out: 110
msp.js:210 MSP data request timed-out: 116
msp.js:210 MSP data request timed-out: 150
msp.js:117 code: 71 - crc failed
MSPHelper.js:1610 CRC error for address 0 - retrying
msp.js:210 MSP data request timed-out: 71
msp.js:210 MSP data request timed-out: 110
msp.js:210 MSP data request timed-out: 116
msp.js:210 MSP data request timed-out: 150
msp.js:210 MSP data request timed-out: 71
msp.js:117 code: 71 - crc failed
MSPHelper.js:1610 CRC error for address 0 - retrying
msp.js:210 MSP data request timed-out: 110
msp.js:210 MSP data request timed-out: 116
msp.js:210 MSP data request timed-out: 150
msp.js:210 MSP data request timed-out: 71
msp.js:210 MSP data request timed-out: 110
msp.js:210 MSP data request timed-out: 116
msp.js:210 MSP data request timed-out: 150
msp.js:210 MSP data request timed-out: 71
msp.js:210 MSP data request timed-out: 110
msp.js:117 code: 71 - crc failed
MSPHelper.js:1610 CRC error for address 0 - retrying
msp.js:210 MSP data request timed-out: 116
msp.js:210 MSP data request timed-out: 150
msp.js:210 MSP data request timed-out: 71
msp.js:210 MSP data request timed-out: 116
msp.js:210 MSP data request timed-out: 150
msp.js:210 MSP data request timed-out: 110
msp.js:210 MSP data request timed-out: 71
msp.js:117 code: 71 - crc failed
MSPHelper.js:1610 CRC error for address 0 - retrying
msp.js:210 MSP data request timed-out: 116
msp.js:210 MSP data request timed-out: 150
msp.js:210 MSP data request timed-out: 110
msp.js:210 MSP data request timed-out: 71
msp.js:210 MSP data request timed-out: 116
msp.js:210 MSP data request timed-out: 150
msp.js:210 MSP data request timed-out: 110
msp.js:210 MSP data request timed-out: 71
msp.js:210 MSP data request timed-out: 116
msp.js:210 MSP data request timed-out: 150
msp.js:117 code: 71 - crc failed
MSPHelper.js:1610 CRC error for address 0 - retrying
msp.js:210 MSP data request timed-out: 110
msp.js:210 MSP data request timed-out: 71
msp.js:210 MSP data request timed-out: 110
msp.js:210 MSP data request timed-out: 116
msp.js:210 MSP data request timed-out: 150
msp.js:210 MSP data request timed-out: 71
msp.js:210 MSP data request timed-out: 110
msp.js:117 code: 71 - crc failed
MSPHelper.js:1610 CRC error for address 0 - retrying
msp.js:210 MSP data request timed-out: 150
msp.js:210 MSP data request timed-out: 116
msp.js:210 MSP data request timed-out: 71
msp.js:210 MSP data request timed-out: 150
msp.js:210 MSP data request timed-out: 116
msp.js:210 MSP data request timed-out: 110
msp.js:210 MSP data request timed-out: 71
````

[image]

mikeller, 2017-11-25:
Thanks for reporting the bug! This is exactly the kind of problems I hoped to be able to identify by doing RC releases for the standalone configurator.

I assume that this is by running a version that you downloaded from the release page and installed (since -RC1 does not seem to be running when installed in Chrome)?

SandroGrzicic, 2017-11-25:
No, I rebased to latest master and manually changed the invalid version tag in the Manifest.js from 10.0.0-RC1 to 10.0.0.

mikeller, 2017-11-25:
Ok, weird then, that used to work.

papayou66, 2017-12-04:
Log work good on RC3 no issue

smitty97, 2017-12-22:
TFSH running 3.15 or 3.23 still cannot download blackbox.  Packet error counter increases approx every 2 seconds

mikeller, 2017-12-22:
@smitty97: What is TFSH?

papayou66, 2017-12-22:
@smitty97 could you send me your log ?

papayou66, 2017-12-22:
@mikeller http://fishpepper.de/projects/tinyFISH

smitty97, 2017-12-22:
@papayou66 the .BBL file it tries to create is 0 bytes.  Is there another log file you mean? I don't have this running in a dev environment,

papayou66, 2017-12-22:
Git is not the best way to solve this issue, join slack if you want here :
http://185.87.186.121:3000/
There is a #support room

ethanzwolf, 2017-12-29:
I just got the Holybro Kopis with the Kakute F4 AIO and have been trying to tune it using blackbox, but for whatever reason, the logs won't save because the "Saving dataflash to file" window just stays at 0% forever. Even small logs (less than 500kb) won't work. I can still clear the blackbox and change its logging rate, but the logs won't download from the onboard dataflash chip. My console is not showing any messages like stated above. Also, once the "download" starts, hitting cancel does not do anything. I have to close BF configurator and relaunch to get it to respond. And maybe this is normal, but CPU usage and cycle time freeze as well, not updating. Thanks for any help. 
I am using the standalone app: Betaflight-Configurator_win32_10.0.0.zip. 
Board Info: 
2017-12-29 @ 12:06:31 -- MultiWii API version: 1.36.0
2017-12-29 @ 12:06:31 -- Flight controller info, identifier: BTFL, version: 3.2.2
2017-12-29 @ 12:06:31 -- Running firmware released on: Nov 7 2017 15:20:35
2017-12-29 @ 12:06:31 -- Board: KTV2, version: 0a

Zethyl, 2018-01-28:
I can confirm the download issue on TinyFish FC using betaflight (non-chrome) 10.1.0

tcalvert, 2018-03-16:
I get the same issue on 10.2.0 -- using a singularity FC running 3.3.0. Just packet error incrementing every ~2sec.

10.0.0-rc3 also exhibits the same symptoms. What workaround are people using?

voneiden, 2018-03-19:
#420 didn't fix it for me. I got a hang every time with nothing else but "MSP data request timed-out". Cancel button didn't work. Upon pulling the USB plug I saw

`serial.js:55: SERIAL: Send buffer overflowing, dropped: 549 entries`

However, I was able to download the log without issues when I removed overclocking from the F4 board (KTV2). Can this be related to OC? @tcalvert are you using OC?

tcalvert, 2018-03-19:
No OC. It just always works on one PC and never on the other.

mazatracker, 2018-04-20:
Same issue here on a new Kakute F4 V2 AIO running 3.2.2 with 10.2.0 configurator. Board info:

MultiWii API version: 1.36.0
Flight controller info, identifier: BTFL, version: 3.2.2
Running firmware released on: Nov 7 2017 15:20:35
Board: KTV2, version: 0

closed 2018-05-27 by stale[bot]