betaflight/betaflight-configurator #749: Filter Display Error: D Term Notch Filter

DimaVIII, 2017-11-26:
Hey Guys,
I found a bug in the configurator.
I set "**D term notch Frequency hz**" to **0** over LUA on my taranis. This actually worked as I was tuning pids (TESTED). Back in Betaflight configurator in the "**Filter Display**" tab it shows me **260** for "**D term notch Frequency hz**". Which is actually wrong. My lua script and the CLI shows me the right value of 0.

Screen Filter Display: http://prntscr.com/hffy00
Screen CLI: http://prntscr.com/hfg3ka

If I turn the slider OFF and add manual **160** for the "**D term notch Cutoff**"  than save and reopen the configurator, it's changing back to **ON** and shows
**260** for "**D term notch Frequency hz**"
 **160**  for "**D term notch Cutoff**"

But only in the  "**Filter Display**" tab not on the CLI.

I think this is somehow related to the ON/OFF slider. ("ischecked" Javascript problem)

basdelfos, 2017-11-26:
Hi @DimaVIII, Thanks for you bugreport. I will have a look at it.

basdelfos, 2017-11-26:
I can not reproduce this bugreport on 10.0.0-rc3.
When I set dterm notch hz in the cli to zero with `set dterm_notch_hz = 0` and `save`. When I look in the PID Tuning tab, the switch for D term notch is set to OFF.

[image]

Which version of the BF configurator did you use? You can see the version under the Betaflight logo.

DimaVIII, 2017-11-27:
@basdelfos you need to switch to another menu like Failsafe, Modes etc. and go back to Filter Display.

I'm using configurator 3.2.2 with Betaflight 3.2.2. (I'm on MacOS)

I can reproduce the issue every time but I made a mistake in my first post, so here again:

- **"Enable D term notch Filter**" slider is **ON**
- set "**D term notch Frequency [hz]**" to **0**
- **SAVE**
- switch to "Modes" menu
- switch back to "**Filter Display**"
and  "D term notch Frequency [hz]" is again 260. 

Only in the UI, the CLI shows "set dterm_notch_hz = 0"

basdelfos, 2017-11-27:
Aha I see, this has been solved in conduction with the changes in PR #735.
This issue has been fixed already and will be released the next configurator release.

closed 2017-11-27 by basdelfos

DimaVIII, 2017-11-27:
This is Offtopic but as you work on this filters and also wrote in the PR (This behaviour is more in line what the firmware expects.). 
If I turn down Cutoff to 0 my Quadcopter gets very sensitive to wind I even lost control completely.
I would like to ask if you experienced the same Flight behavior like me?

basdelfos, 2017-11-27:
You mean setting the cutoff to 0 and hz to a value higher than 0? I have to look it up, but my guess is that the filter is also inactive when cutoff = 0. If you have a noisy D then this would cause oscillations and even hot or burned motors.

DimaVIII, 2017-11-27:
Yes cutoff and hz to 0.

Short: Chamelon on T Motor F40 III 2400kv, DShot 1200
My quad was on “stock” setting crazy about wind, just a bit wind and it turned upside down the also motors got quit hot (but still ok).
After “I term” tuning it was fixed even heavy wind was not a problem anymore. Than I went to the PT1 filters and disabled the last two filers (D term notch hz and cutoff filter) and the quad was again oscillating in wind so hard that it turned upside down and crashed.
So I turned cutoff back to 160 and left hz at 0. This fixed the wind oscillations and the quad was flying really almost perfect. 
Also after I set cutoff 160 and hz 0 the motors been way cooler than in the whole tuning before. It seams for me like this setting triggerd something magical xD.

I saw that a very good Chameleon Pilot used this setting (Suki FPV https://youtu.be/wNl1DA-yNNw from around 20min), he also have very high “I terms” like me.

mikeller, 2017-11-27:
@DimaVIII: Could it be that the other pilot uses dynamic filtering, and you don't?

DimaVIII, 2017-11-27:
@mikeller I activated dynamic filtering.