betaflight/betaflight-configurator #753: Resets Arduinos

LightningStalker, 2017-12-01:
Betaflight-configurator just reset my Arduino on a different USB port that was doing something VERY IMPORTANT. All I did was save and reboot under the configuration tab and the Arduino reset and the tty of my arduino was selected.

mikeller, 2017-12-01:
That's unfortunate. We do probe all ports to find connected flight controllers, which I think is in the best interest of most users.
If your arduino is doing important things, and is triggered to reboot when being probed, then I would recommend you do not connect it to a workstation while doing important things.

closed 2017-12-01 by mikeller

LightningStalker, 2017-12-02:
I would definitely do that if I could, but it must be reprogrammed nearly every day.

mikeller, 2017-12-02:
Unplug it when not programming it. If you're lazy (like me), get yourself an USB hub with switches: https://www.aliexpress.com/item/Multi-7-port-USB-Hub-2-0-Adapter-7-Ports-Hub-USB-On-Off-Switch/32816240502.html

LightningStalker, 2017-12-02:
I'm going to unplug it, take the cable, wrap it around my neck, and use it to hang myself.