betaflight/betaflight-configurator #757: Maximum ARM Angle disappeared

spuder, 2017-12-04:
In 10.0.0-rc3 the arming section is empty

[image]

Same quad works on 3.2.2

[image]

It is entirely possible that it was moved, and I can't find it.

mikeller, 2017-12-04:
Weird thing, I get the 'maximum arming angle' input on 10.0.0-rc3. What version of the firmware are you testing with? Could it be that it's too old to support UI entry of the maximum arming angle (the fact that it's showing an empty input field on 3.2.2 kind of suggests so).

spuder, 2017-12-04:
Running 3.2.0 

    # Betaflight / BETAFLIGHTF3 (BFF3) 3.2.0 Sep 11 2017 / 06:30:07 (649ccd7c7) MSP API: 1.36

I can't remember if that was the 3.2.0 build that I compiled myself or not, so I flashed with the official 3.2.2 build and restored configs. I see the same result. 10.0.0-rc3 has missing Maximum ARM angle, and is present on 3.2.2 (Why the field is empty is another issue I'll need to look into)

mikeller, 2017-12-04:
Mystery explained: It's only supported from 3.3 onwards.

spuder, 2017-12-04:
Ah yes 🤦‍♂️.  Completely forgot about that.

closed 2017-12-04 by spuder