betaflight/betaflight-configurator #761: [Feature Request] Translation to multiple languages

McGiverGim, 2017-12-04:
Some users asked to translate the configurator. I did that for the Cleanflight Configurator, so I've created a collaborative web for the Betaflight Configurator in:

http://betaflight.oneskyapp.com

Where people can help to translate it in an easy way. If some needs that I add a new language, please told me and I can add it.

Some things:
- Some words (for example pich, roll and yaw) maybe are better understood in english and is better to not translate them... or not. Each language can decide if translate them or not, suggestions from the users of each language are welcome. It can be discussed here, or in the oneskyapp web (it has a forum).

Integrating the changes: 
- From time to time the base of the translation must be updated to be in line with the code in github.
- Before making a new release, the translations must be incorporated in the code. 
I can do both things, but it will be great if some "official developers" decide this. I can give them administrator access. There's an API too to access "automatically" to the translations, if needed.

McGiverGim, 2017-12-05:
Until we found a better way, I will try to update regularly this branch:

https://github.com/McGiverGim/cleanflight-configurator/tree/bf-translations

In this branch I will try to have the have the latest version of Betaflight Configurator and the latest version of translations. In this way the translators can "test" his job.
If some translator need that I update it at certain moment, please tell me and I will do as soon as I can.

Thanks!

mikeller, 2017-12-05:
@McGiverGim: Just a few thoughts from me: Taking into consideration that the texts in cleanflight configurator and Betaflight configurator are > 95% identical, it would probably make sense to first put some effort into coming up with a merged translations file, so that translations can be generated / maintained for both cleanflight and Betaflight at the same time.

What we really need to for this to happen is:
- get a diff between the two translation files;
- whereever possible, if the only difference in a string is 'cleanflight' / 'Betaflight', use string formatting to set the firmware type;
- for other cases (like help texts that are different, but have the same name), use a helper function (`getI18n(name)`) to get a the translation by appending the firmware type to `name`.

Thoughts?

McGiverGim, 2017-12-05:
I thinked too that the two files will be almost identical, but now they aren't. The CF one is older, so maybe when @hydra merges the latest updates, maybe it will be.

Other thoughts: the difference is not only the CF/BF word, there're others:
- Styles: some styles are inside the messages file, to use CF or BF colors. I think that this is not a good idea, is better to have these differences in a CSS and use the same class style in the messages file.
- URLs and other strings: the same message but different urls or texts (web page, docs page, donations, etc.)

The effort can be made, but there's a lot more than a 5% of the strings. 
The CF and BF configurators will continue being very similar in a future? What will be the way? Made the effort in BF and then, when @hydra updates CF then update the strings needed?

mikeller, 2017-12-05:
Yeah, the separation of styling and text could be improved as well. But in a lot of cases it is styling of individual parts of a message, and getting that completely out of the text will involve a lot of extra effort, so not sure if it's worth the effort.

The URL situation would have to be solved with `getI18n(name)`.

And yes, as long as cleanflight itself is 98% based on Betaflight, the configurators will likely stay this similar as well.

McGiverGim, 2017-12-05:
What about this? To simplify, if the i18n of chrome let's do it as this:

Split the actual messages.json into two files: the actual messages.json and other new, custom_messages.json (or the name we decide).

In the `messages.json`, we can let the strings like this:
- Strings that are the same in CF and BF.
- Strings that only differ in some style, moving the style to the CSS file. The major part are changing the `class="message-possitive"` to `style="color: #ffbb00"` so with a search and replace I think it will be enough and will not be too much effort.
- Strings that only differ in the word `Cleanflight` for `Betaflight`, changing this for a parameter that ideally will be readed from the `custom_messages.json` file. This can be done too with a search and replace.

In the `custom_messages.json`:
- Any other string that is different (it has different text, urls, etc.)

In this way, the message key remains the same than now, so we don't need to change the actual translations (only move from one file to another and made some automatic replacements) and we don't need to change almost nothing in the code of the configurator.

But I need to test if the Chrome i18n api lets do this.

The only problem is to detect strings that by now are only in CF and not in BF, if there're any, when merging this into CF. But in the worst case, a custom made program can do the job (or maybe a linux command, if some of you control this). 

What do you think?

mikeller, 2017-12-05:
- not sure how the Chrome / NW.js support for i18n works, as in if there are multiple files in a folder for a language, will they get merged?
- anyhow, I don't particularly like it, since:
- it will make editing translations harder for translators, since they now will have to manage 3 files instead of one;
- there is an overhead for coordinating where individual strings go, and moving them between files if their content changes.

But agreed that strings that only differ in styling between cleanflight and Betaflight should definitely be addressed by moving the styling into CSS.

McGiverGim, 2017-12-06:
I've been taking a look to this. It seems that Chrome is the only API/framework that does NOT admit more than one file, so it seems that my solution is not possible (at least not directly) :( 

The translations for the translators are not a problem. They have a web page, with projects to translate and is very easy for them select one and start translating. But as I said, it seems that with chrome is not possible.

In the other hand, with your solution, I think it will work, but I don't know how much work will be. Maybe to simplify it we can let the same function for all strings, the `getI18n()` and this function will search for the `name` and if not found search for same `name` but appending the firmware type. In this way all the code will use the same function.

The thing that I don't like of this solution is:
- We will have strings for cleanflight in the betaflight project and strings for betaflight in the cleanflight project.

And another problem (independent of the solution), how this will work when working? We need a `master` messages file. For example:
- CF makes a change, will be included in BF in a future.
- BF makes a change, will be included in CF in a future.
- What file must we upload to the translators repository? The BF one? The CF one? Someone will make a merge manually and then make the commit to the two repositories, with the problems that this can give? The easiest solution will be take one as master (BF for example) but in this way CF will not have the translation ready until the new code is merged into BF.

I think we need to solve this problem before making any change.

mikeller, 2017-12-06:
Yeah, the mergeing is still unresolved. We could also go for a solution where we use scripts to merge the two translation files for Betaflight and cleanflight into one, and then split the resulting translations into two again. Or do it simpler, and get all of the Betaflight file translated, and use a script to only extract what is different in cleanflight and get that translated separately, and use the rest from Betaflight.

There seems to be a lot of motivation and movement going with respect to translations right now, so I'd like to harness that and get translations (albeit imperfect in format) while it is going. We can do a cleanup later - it will be more work than doing it first, but way less work than finding people that are motivated and capable to help us with the translations.

McGiverGim, 2017-12-06:
Yes, make the cleanup later will be not a problem. The code changes need to be made now or later, and the translations can be updated automatically, I can make a program that compares english before and later the cleanup and updates all the translations in the same way, to not loose effort (and motivation) from the translators.

McGiverGim, 2017-12-06:
Only to let some data, to translate the Betaflight file, I started from the Cleanflight one because the work for the spanish translation was done. The differences between both files were:
```
124 New  66 Changed  21 Deprecated
```
The total Betaflight messages file has 923 phrases. 124 are new, 66 are different and 21 are missing from the CF version. So the difference at this moment between both are more or less the 20% of the total. 

I'm pretty sure that the changes in the major part can be solved automatically extracting the style to a CSS and including the Betaflight word as parameter, but only as a reference.

McGiverGim, 2018-01-01:
The translations have been merged, so I close the issue. Thanks to all!

closed 2018-01-01 by McGiverGim