betaflight/betaflight-configurator #762: Multiple presses of adjusting arrows are glitchy

llambkin, 2017-12-04:
Description of problem:
I press the adjusting arrows to adjust PIDs, and if I press it multiple times, sometimes, it presses down, sometimes it just stops adjusting.

mikeller, 2017-12-04:
What version of the configurator, and what operating system are you running on, @llambkin?

richard-scott, 2017-12-05:
I get this to on a Windows 10 system using the latest configurator code checked out from GitHub. I tried to record it using CamStudio, but strangely enough when recording the BF Configurator Window the issue isn't there. Upon stopping recording and clicking on my mouse button multiple times wwithout moving the mouse pointer the input jumps focus from up number arrow to down number arrow changing the value accordingly either up or down. Oh, this also is an issue with different flight controllers and does the same in the new Standalone configurator app too.

mikeller, 2017-12-05:
@richard-scott: So it happens with the non-standalone (i.e. Chrome) version as well?

richard-scott, 2017-12-06:
@mikeller Yes, it happens in both variants of the Configurator. I'll try and get a video via my mobile phone, as recording on the same computer seems to stop the issue happening. I will try after work today.

richard-scott, 2017-12-07:
[Here](https://drive.google.com/file/d/1Z7nRP4EzlLBCzbehEf9sFsrYcbf6MEAg/view?usp=sharing) is the link to the video I took. The mouse cursor does not move, and if you turn up the volume you can here my mouse clicks. You can see that the input focus jumps about. This time it isn't changing the number in the other direction when focus changes, I've not been able to reproduce that.

klutvott123, 2018-11-22:
I'm experiencing this too with standalone and chrome version on windows 10 on two computers. Doesn't the standalone version use some chrome components too? 

Clicking the up arrow multiple times will eventually cause the down arrow to be clicked(but seems like no mouse event is generated) even if the cursor hasn't moved. 

Clicking either arrow multiple times will cause the input box stop registering the clicks. When this occurs, a mouse event is generated when moving the cursor away from the arrow buttons(without clicking). 

Maybe it is something similar to what is described here:
https://stackoverflow.com/questions/34282278/arrows-not-working-on-input-type-number

Someone posted this https://jsfiddle.net/MCAU/Lpojfrm2/ in the above link. Running it in firefox works, but in chrome it doesn't. When clicking the up arrow, the down arrow is clicked instead. Changing 'mousemove' to 'mouseup' makes it work in chrome too.

I also read something about chrome now using pointer events instead of mouse events. Is it possible that this problem has something to do with mouse events? 

Chrome users have also experienced clicks not being registered when cpu usage is high. 
Betaflight configurators cpu usage is also relatively high. Probably because of it continuously receiving rc data from the flight controller. This has to generate a lot of events too. Maybe there's just too many events. Or do we just have too slow computers? 😆 

@mikeller any thoughts on this?

mikeller, 2018-11-23:
Well possible that this isn't working reliably. I am no expert in the UI componentry we are using in configurator, but it is way old and could / should probably be replaced with newer versions / a newer framework, if somebody was to step up and take on the tedious work to make this work.

klutvott123, 2018-11-23:
I would do it if i could, but this is above my skill level. My html/css/js knowledge is extremely limited. 

I did do some experimenting. I tested it on linux and it seems to work just fine there. I also tried reducing the rate at which the configurator requests rc data from the fc. Brought the port utilization down to 2%, but didn't make much difference so i think we can rule that out. Google tells me that a lot of people are experiencing the same with chrome/chromium. https://bugs.chromium.org/p/chromium/issues/detail?id=740237. 

Seems like the problem only affects ``input type="number"``. One solution could be to replace the input with something like a jQuery spinner. I tried placing a regular input and a jQuery spinner at the top of the pid page. The regular input exhibits the same behaviour as the rest of them, but the jQuery spinner works just fine. But it's not exactly a drop in replacement.

klutvott123, 2018-11-25:
I said in my above post that it seems to only affect input type "number", but it affects all input types that use the spin buttons. 

I'm able to reproduce with the below code in chrome:
```
<!doctype html>
<html lang = "en">
	<head>
		<meta charset = "utf-8">
		<link rel="stylesheet" href="https://code.jquery.com/ui/1.10.4/themes/smoothness/jquery-ui.css" />
		<script src="https://code.jquery.com/jquery-1.10.2.js" /></script>
		<script src="https://code.jquery.com/ui/1.10.4/jquery-ui.js" /></script>
		<script>
		$(function() {
			$("#spinner").spinner();
		});
		</script>
	</head>
	<body>
		<div>
			<input type="text" id="spinner" min="0" step="0.01" max="1" />
		</div>
		<div>
			<input type="number" min="0" step="0.01" max="1" />
		</div>
		<div>
			<input type="date" />
		</div>
	</body>
</html>
```
I added a jQuery UI spinner at the top there to compare with. 
If i move the mouse just a little bit between each click it works perfectly, so i'm thinking maybe it can be "fixed" by firing a mousemove event manually on each click. 

@ProcessDestroyer Scrolling seems to be a little buggy too. When starting the configurator i have to do some serious clicking before it starts working. 😆

klutvott123, 2018-12-01:
@Benjamindudok Which operating system are you using? This seems to be a problem with chrome on windows 10(possibly other windows versions). Works fine in firefox here, but not in chrome. I have done some investigating in the debugger and found that after clicking a spin button a few times triggers a mouseleave event. Then you have to move the cursor just a little bit to make it focus on the input again. It's much worse in betaflight configurator than it is in the chrome browser for some reason. I think there is a bug in the way chrome handles mouse clicks on the spinbuttons, and from googling i found that people have been experiencing similar things for years. If it is, it probably doesn't have a high priority since how often do people use spin buttons in browsers? 

One possible solution is to hide the spin buttons and apply a jquery ui spinner to all inputs of ``type="number". I did it for one of them, but couldn't get the input events to fire when changing the value using the spinner. Then i took a break. 😄

Benjamindudok, 2018-12-02:
I'm using windows 10 and the latest chrome browser (70.0.3538.110). I am now able to reproduce the issue outside the configurator too. The issue does get really noticable when javascript is busy calculating stuff. See this [jsfiddle](https://jsfiddle.net/BenjaminWD/79tfnj4e/13/), where i put an interval at around 33fps  (like the get_fast_data function is doing in the configurator) and only then the issues starts happening (for me atleast).

I'll see if i can easily add some custom buttons to all the number inputs which shouldn't be to hard to implement, or else use the jquery ui spinner.

klutvott123, 2018-12-05:
@Benjamindudok You're right. It's much easier to reproduce in chrome with the jsfiddle you posted. Let us know if you manage to do this. I would be happy to test if needed.

Benjamindudok, 2018-12-08:
Sadly i'm not able to provide a (temporary) solution to fix this issue. it's quite difficult to create custom spin buttons while providing the same functionality (mousewheel scrolling for example, when you hide the native spin buttons, you also prevent mousewheel events). And adding jquery-ui spinners summons a host of other issues besides adding a ton of needed styling.

I did find an official bug report on this in the chromium project, and added a comment with my findings and a link to this issue too. Hopefully they can reproduce it, and provide a fix for the native input.

https://bugs.chromium.org/p/chromium/issues/detail?id=908252&can=2&start=0&num=100&q=input%20type%20number%20spinner&colspec=ID%20Pri%20M%20Stars%20ReleaseBlock%20Component%20Status%20Owner%20Summary%20OS%20Modified&groupby=&sort=

klutvott123, 2018-12-09:
@Benjamindudok Yeah, your're right. I noticed the mousewheel stops working too, but that seems to also be buggy(at least for me). If the chromium devs manage to fix the input, that would be ideal. At least with the code you provided they should have no problem reproducing it. 😃

danybd, 2021-03-24:
I tried in windows 10 with the 10.8.0 and can not reproduce the described issue. Is it still an issue at all?

haslinghuis, 2021-03-25:
Guess the issue should be closed as there was no activity for over a year. If the issue should arise again we can always create a new one or reopen.

klutvott123, 2021-03-25:
Just tested in windows 10 with a recent nightly build. can't reproduce anymore. Someone did something that fixed it at some point. @haslinghuis was it you? 😄

danybd, 2021-03-25:
> Just tested in windows 10 with a recent nightly build. can't reproduce anymore. Someone did something that fixed it at some point. @haslinghuis was it you? 😄

That´s great. One bug less 😄

haslinghuis, 2021-03-25:
😇

mikeller, 2021-03-28:
I **love** closing stuff that's fixed itself. :laughing:

closed 2021-03-28 by mikeller