betaflight/betaflight-configurator #771: Windows Protected your PC Error

richard-scott, 2017-12-06:
Are there any plans to get the required certification to remove this error?

[image]

mikeller, 2017-12-06:
I wasn't aware that this is showing at all. I run Windows defender, but it does not show for me. Will look into it, but not too keen to pay racketeering money to Microsoft for them to stop harassing the app. Also, this will make automated builds on CI difficult.

dthubereng, 2017-12-06:
Happened to me this morning. Not a big deal to hit "Run anyway" the first time, IMO.

fixmania, 2018-01-17:
In my case, that is appearing only in Windows 10 that has more powerful PC protection. But only about when you install the app.

closed 2018-05-13 by stale[bot]