betaflight/betaflight-configurator #772: New Standalone Configurator doesn't remember last used size/location

richard-scott, 2017-12-06:
It would be great if the New Standalone Configurator could remember the last used window size and location. For example I like to have my Configurator running maximized on my screen. Each time I exit and load the standalone configurator it doesn't restart maximized but it uses its own pre-set window size.

basdelfos, 2017-12-06:
I've found an easy drop-in-place solution for this. Should also be implemented in the blackbox and blheli configurator standalone app.

https://github.com/nwjs/nw.js/wiki/Preserve-window-state-between-sessions

mikeller, 2017-12-06:
@basdelfos: This even exists as an npm module: https://www.npmjs.com/package/win-state
What do you think about adding it from there, instead of adding a static copy?

basdelfos, 2017-12-06:
@mikeller great, didn't look further when I quickly did google search. The npm module is definitely more preferred way to go.

dzenbot, 2018-01-05:
Shouldn't this be closed as fixed?

basdelfos, 2018-01-05:
Yes this is fixed with the merge of #782.

closed 2018-01-05 by basdelfos