betaflight/betaflight-configurator #774: Building with the new node.js system

McGiverGim, 2017-12-07:
I've tried to execute it under windows, installed the latest version of `node.js`, execute a `npm install`without error, but after that, when executing the `npm start` I get an error, this is part of:
```
19 verbose node v9.2.0
20 verbose npm  v4.5.0
21 error code ELIFECYCLE
22 error errno 1
23 error betaflight-configurator@10.0.0-rc4 start: `node_modules/gulp/bin/gulp.js dist && node_modules/nw/bin/nw .`
23 error Exit status 1
24 error Failed at the betaflight-configurator@10.0.0-rc4 start script 'node_modules/gulp/bin/gulp.js dist && node_modules/nw/bin/nw .'.
```
Finally I've reach it to work doing two things:
- Executing this command `npm install nw` after `npm install` (it seems a dependency is missing somewhere?)
- Change this line in the package.json:
https://github.com/betaflight/betaflight-configurator/blob/040757b74e0c626f92eb929ce1f49b204f54dd74/package.json#L8
For this other:
```
"start": "node node_modules/gulp/bin/gulp.js dist && node node_modules/nw/bin/nw .",
```
I don't know if this is a problem of my system or not. Some idea?

basdelfos, 2017-12-07:
Yep this is known by me since yesterday. `npm start` is the old way to start the app. New way is to use `gulp apps`, `gulp debug` or `gulp release`.
I will remove the old way from the readme.

McGiverGim, 2017-12-07:
Yes, that worked. I close the issue. Thanks!

closed 2017-12-07 by McGiverGim

mikeller, 2017-12-07:
Sorry guys, that was my bad, I removed the 'nw' devDependency when removing unneeded dependencies.

PR #778 (2017-12-08): Fixed missing 'nw' devDependency.
Fixes #774.