betaflight/betaflight-configurator #788: Not Entry Point Found Error - Betaflight Configurator 10.0.0

RoughRiderFPV, 2017-12-13:
Hello, My issue is stemming from a startup error. Error Window Entry Point Not Found says "The procedure entry point QueryUnbiasedInterruptTime could not be located in the dynamic link library KERNEL32.dll." I've Redownloaded this file 4 times and am stuck at the moment. What should I do?

mikeller, 2017-12-13:
What platform are you running, and on what hardware?

RoughRiderFPV, 2017-12-13:
windows 7 32bit

mikeller, 2017-12-13:
I could not find any information about Windows 7 being supported by NW.js, but the error message, and the fact that support for Windows 7 has ended, suggest that might not be supported any more.

You'll probably have to update your operating system to one that is supported, or look into building the configurator with a legacy version of NW.js.

closed 2017-12-13 by mikeller

basdelfos, 2017-12-14:
I have a Windows 7 VM (64-bits) laying around, will try tonight if I get the same error.

basdelfos, 2017-12-15:
Confirmed that the configurator does work on Windows 7 64-bit, I'm running SP1, probably very old. I did some testing and everything seems to work correctly.

mikeller, 2017-12-15:
Ok. That still does not mean that it runs on Windows 7 32 bit. Google has been rather progressive in dropping support for older platforms in Chrome.

basdelfos, 2017-12-18:
I have no Windows 7 32-bits available to test

mikeller, 2017-12-18:
Might be hard to find these days.

mikeller, 2017-12-27:
Duplicate of #813.