betaflight/betaflight-configurator #793: Provide access to unstable releases.

richard-scott, 2017-12-16:
When selecting to flash a firmware and having the "Show unstable releases" enabled, it would be great if the BFC could allow us to install the latest release from here:

3.2.x release: https://ci.betaflight.tech/job/Betaflight%20Maintenance%203.2/lastSuccessfulBuild/artifact/obj/
3.3.x release: https://ci.betaflight.tech/job/Betaflight/lastSuccessfulBuild/artifact/obj/

As long as that part of Jenkins hasn't crashed (as it has done in the past) this URL should be reliable for HEX file downloads.

AndersHoglund, 2017-12-16:
No. There are three public code maturity levels, and I think we should keep it that way.
1. Releases, available directly in BFC flasher.
2. Release Candidates, available as "unstable" in BFC flasher.
3. Experimental Daily builds, available from the CI site. Install if you dare.
4. Private development debug builds. 

With your suggestion no 2 would disappear, or merge with no 3.

richard-scott, 2017-12-16:
No, with my suggestion you'd also have a no 3 alongside 2 and 1.

mikeller, 2017-12-17:
I think this is a good idea, adding a 'Latest nightly build' option (like many other software projects do). The only concern is that increased traffic could be a problem for the server. @wind0r, what is your take on this?

richard-scott, 2017-12-18:
If you are concerned about load on the CI server, get the  CI server to upload the latest artifacts to a GitHub repo or to Dropbox or something. Let someone else's site take the load.

wind0r, 2017-12-18:
Server should handle that fine, but I dont realy know if we want that, mergeing should be a little more strict then and if it gets added it should be hidden and there allways should be a warning popup!

mikeller, 2017-12-18:
@richard-scott: 

> If you are concerned about load on the CI server, get the CI server to upload the latest artifacts to a GitHub repo or to Dropbox or something. Let someone else's site take the load.

That would take extra effort to add an API to GitHub / Dropbox to allow the configurator to query and list the latest build. Allowing the user to flash a 'mystery build' of unknown vintage and age is not going to be an improvement.

@wind0r: Agreed about the warning. My idea would be to change the 'Show unstable' for a dropdown with choices of 'stable' (default) / 'stable + release candidates' / 'stable + release candidates + nightly builds', with an explanation / warning next to it. The warning should be such that we would not have to change our development process because of it - i.e. tell the user that there is not expectation that the nightlies even fly.

McGiverGim, 2018-07-02:
This feature request is already implemented in the configurator. I close it.

closed 2018-07-02 by McGiverGim