betaflight/betaflight-configurator #797: not running on mac os 10.7.5

macht-nix, 2017-12-17:
my field laptop (early 2008 MacBook) only runs 10.7.5 (f**k apple) 
please don´t force me to buy a new computer

basdelfos, 2017-12-18:
Maybe PR #799 can help you not needing to buy a new mac

basdelfos, 2017-12-18:
@macht-nix sad news. Older macOS versions are not supported in nw.js. We don't want to include builds that are not officially supported anymore as this could lead to support questions we can not answer or even not fix.

But not to let you behind with an unuseful field laptop, I managed to build the configurator by overriding the nw.js version to the last stone-age version that supports osx32 ;-). It build successful, but can not test run it myself (doesn't run on a recent macOS version). I hope it runs on your laptop, unfortunately we can not give any support for it.

You can download it here:
https://drive.google.com/file/d/1VkkcPy5uc5V7-yT90AXZX_OrQIQPindU/view?usp=sharing

closed 2017-12-18 by basdelfos

macht-nix, 2017-12-18:
@basdelfos first thank you for trying to help me

i tried it and it just opens as a safari browser window and is not seeing any serial ports or showing the app content

...

mikeller, 2017-12-18:
@macht-nix: That's what we suspected. The version of NW.js that runs on osx32 is so old that it does not support the APIs that we need for configurator (like serial port access).