betaflight/betaflight-configurator #798: Resizing window smaller can make the save and reboot button disappear.

CrazyMonkeyCM2, 2017-12-17:
Resizing window smaller can make the save and reboot button disappear.
Should not be possible to do so.

mikeller, 2017-12-18:
What tab is this happening for, @CrazyMonkeyCM2?

CrazyMonkeyCM2, 2017-12-18:
Sorry, should have mentioned, all of them.
I am running my screen at 3840x2160 with DPI scaling at 350%.
Also its only the window width that causes it, not height.

mikeller, 2017-12-18:
- there isn't a 'Save and Reboot' button on all tabs;
- tried with one, works fine.

Closing this as 'not enough information'.

closed 2017-12-18 by mikeller

CrazyMonkeyCM2, 2017-12-18:
Ill provide pictures if you don't believe me.
here is it happening on 2 tabs, but it happens on every tab save is supposed to show.
[image]
[image]

[image]

[image]