betaflight/betaflight-configurator #805: ARMing doesn't activate exclamation mark

richard-scott, 2017-12-18:
I'm testing the [PREARM ](https://github.com/betaflight/betaflight/issues/4756) fix and I've noticed that on build 430 from the CI server and the ARMSWITCH gets activated, but the exclamation icon doesn't light up. I've tried BFC 10.0.0 and 10.0.1.

The CLI status shows ARMSWITCH but the icon doesn't light up.

McGiverGim, 2017-12-19:
I've looked at it, the "problem" is the `bit_check`:
https://github.com/betaflight/betaflight-configurator/blob/bd1545484d888222dab4a15df91f461b3f272b23/js/serial_backend.js#L544-L547
I suppose the `CONFIG.mode` is not ok. I will try to find the problem, but I'm not too sure if my skill is enough for this ;)

McGiverGim, 2017-12-20:
@richard-scott when you say that the ARMSWITCH gets activated, are you talking about the MODES tab and the ARM box?

richard-scott, 2017-12-20:
@McGiverGim No, i'm talking about the icon on the top nav bar near to battery voltage. There is an exclamation icon on there that normally gets activated when the FC is in "Armed" status.

richard-scott, 2017-12-20:
This one:

[image]

McGiverGim, 2017-12-20:
Yes, but I was talking about:
>>>and the ARMSWITCH gets activated

that you commented. If I'm not wrong, you were talking about the `status` command, that shows `ARMSWITCH` but if I'm not wrong, nor the icon on the top, nor the mode with ARM are lighting when armed. Right?

McGiverGim, 2017-12-20:
It seems a Betaflight firmware problem. What is the date of the build 430?

I've seen a recent commit that maybe solves it. I will try to update in my build and test it.

richard-scott, 2017-12-20:
When flicking my ARM switch on my Transmitter I get the following in BFC:

1. The CLI Status command show the ARMSWITCH status.
2. The Modes Tab ARM box goes yellow.
3. The exclamation icon next to battery voltage does NOT illuminate.

And yes, the BFC stopped working when this was merged:
https://github.com/betaflight/betaflight/issues/4756

McGiverGim, 2017-12-20:
I think that maybe https://github.com/betaflight/betaflight/pull/4778 solves the problem, maybe your firmware is between both revisions?
I will try to test it as soon as possible.

richard-scott, 2017-12-20:
From looking at the change log I had been using build 430, but looks like I need to try build 435 or newer:

```
#435 (18-Dec-2017 19:43:07)
Add conditionals for MCU dependent ADC drivers (commit: 3b3ed72) — shinoda / githubweb
Mask BOXARM in MSP enables (commit: 5048c74) — lounatic / githubweb

#430 (18-Dec-2017 02:33:10)
MSP mode enable refactor (commit: 8b4bfc3) — lounatic / githubweb
```

I may be able to test tonight, but should be able to before the end of the week.

McGiverGim, 2017-12-20:
I've tested it and it works :)

So this bug is solved. Test it at home and the close the issue if all is ok ;)

richard-scott, 2017-12-22:
I've just tested with build 445 in the 3.3 branch this looks to be fixed :-)

closed 2017-12-22 by richard-scott