betaflight/betaflight-configurator #807: Window name not defined

McGiverGim, 2017-12-20:
Some days ago I created this PR because the icon was missing in taskbar/window:

https://github.com/betaflight/betaflight-configurator/pull/791

I've observed that was removed in this other PR:

https://github.com/betaflight/betaflight-configurator/pull/782

This other PR removed other info, like `title` (now it shows `betaflight-configurator` and not `Betaflight Configurator`). I've tried to add it again without luck, the title doesn't change. Any idea?

mikeller, 2017-12-21:
Yes. If you look at #782, what this does is essentially close the main window before it's showing, and instead of this have `eventPage.js` open a new window. The benefit of this is that this new window will persist its size and position, unlike the NW.js generated main window.

You'll have to look into adding the title and icon to the new window. @basdelfos can probably help with this.

McGiverGim, 2017-12-21:
Yep, yesterday I didn't found the parameter in the chrome app documentation, but I've tested to add a new parameter called `title` and... it worked :)

I will push the PR now...

basdelfos, 2017-12-21:
Yes, PR #782 closes the nw.js window and uses the window that is opened in `eventPage.js`. This has advantages like automatic saving of window size, state and position (and also some other events that get caught). I'm on a mac and didn't notice that the Windows title had been lost.

Ah, great you just posted you found a solution! Great job!

closed 2017-12-22 by mikeller