betaflight/betaflight-configurator #813: Beta-flight config system requirements?

LlamaB, 2017-12-21:
Heyo, can't find any mention of this off hand. Is the configurator compatible with windows XP?

I was recently given a old laptop that would be nice to take to field and let everyone use to tune.  Given the old hardware and cost of a new OS it's going to have to stay on XP.

When trying to boot up beta flight config on it I get the following error:
``The procedure entry point ReleaseSRWLockExclusive could not be located in the dynamic link library KERNEL32.dll``

The computer has not booted since 2005 and I am still working on cleaning up the junk, getting it connected to net and updating as far as XP will go.

Any tips or confirmation whether I will ever make it work or not would be appreciated, thanks.

mikeller, 2017-12-21:
Sorry to be the bearer of bad news. But there is no chance this will ever work on Windows XP, and little chance that the configurator will work on any Microsoft operating system that can run on this hardware (it does not run on Windows 7 32bit). You might be able to get lucky by installing linux on this machine, the linux32 version of the configurator should then run on it.

LlamaB, 2017-12-21:
Ok, thanks.

closed 2017-12-21 by LlamaB