betaflight/betaflight-configurator #817: New Configurator 10 Strange Behavior

claytonbakerjr, 2017-12-24:
Tonight I was flashing one of my boards to BF 3.2.3
Everything seemed to be OK until I placed my DIFF file back into my configuration via the CLI.
Once I entered in everything and typed and entered Save, I could see it simply state: saving in the CLI.
So thinking it was complete I typed exit and nothing happened, it didn't even show exit in the CLI after hitting enter.
So I ended up clicking Disconnect and then I reconnected to the board and verified that the save indeed completed. I then changed one more thing in the Configuration via the Configurator and clicked Save and Reboot....nothing happened...all I did see up top was it stated EEPROM Saved. But the typical disconnect and reconnect did not happen at all. So again I clicked Disconnect and then reconnected to the board. The change I made was saved.
I tested this on a total of 4 quads with the following boards
CLRacing F4 (version 1) Flashed with CLRACINGF4 3.2.3 and even reflashed back to 3.2.2
CLRACING F4S flashed with CLRACINGF4 3.2.2 (I have 2 of these boards)
Hobbywing XROTOR F4 flashed with OMNIBUS F4SD 3.2.2
In every case I can repeat the following
Connect to each
Go into the CLI type version and see it complete command and show version of firmware.
Type Exit, nothing happens the screen stays unchanged.
In the Configurator itself in each case if I change anything and click Save and Reboot the GUI does nothing.
With the Hobbywing board when in the CLI or the Configurator via the buzzer I can here the INIT
sound off however the CLI and/or the Configurator GUI does not respond or change.
In all cases I have to manually click Disconnect and then Connect to get back into the board.
Up until tonight I was using the older Chrome Configurator and decided to switch over to the new executable version.
I also just checked this issue against the older Chrome version and it works just fine. The commands respond like they should within the Chrome GUI
Please let me know if you need anything else and if this is mentioned elsewhere please let me know.

claytonbakerjr, 2017-12-24:
BTW to add I tested 3 different USB cables in the event it might be a comms issue

claytonbakerjr, 2017-12-24:
Well interesting...after closing the new Configurator (which I did several times to including rebooting PC) and as mentioned used the Chrome version to check upon closing the Chrome version and re-opening the EXE version things seem to now be working. I'm scratching my head on this one.
I will keep an eye on this for a day or so since I do have several other things to flash and setup over the holidays,

closed 2017-12-27 by claytonbakerjr

richard-scott, 2017-12-28:
I'm about to open a new case as my CLI usage hangs the standalone configurator when leaving the CLI tab.
Maybe your not that mad after all.

Connect to an FC. click in CLI (do nothing, or do something doesn't really matter) click out of CLI and BFC hangs.