betaflight/betaflight-configurator #818: Update warnings on firmware flasher page

mawais15, 2017-12-25:
Some warnings on firmware flasher still showing chrome related stuff like "If you have problems flashing try disconnecting all cables from your FC first, try rebooting, **upgrade chrome**, upgrade drivers"
I think upgrading chrome is no more required because its a standalone application now.

mikeller, 2017-12-25:
Thanks for reporting this. We are still in the process of ironing out the details of the switch to a standalone application, and we will be amending the documentation when we get to it.

(Also, take note that the Chrome web store version of the configurator will be maintained for now, since it still works on Chromebooks, which some people use to configure their craft while on the airfield.)

danybd, 2021-03-16:
@mikeller Issue outdated? Should be closed as no more chrome support?

mikeller, 2021-03-16:
@danybd: No, in fact the messages referring to Chrome are still in the application and by now they should be removed:

[image]

PR #2451 (2021-03-28): Remove chrome references
Fixes: #818 by removing references to chrome and checks for ChromeOS.
closed 2021-03-28 by mikeller