betaflight/betaflight-configurator #821: Feature Request: Only display features in BFC that are compatible with the FC version in use.

richard-scott, 2017-12-28:
To help with future releases of the firmware that are incompatible with the Configurator it would be great if every new feature had a "minimum firmware version" tag in source code was only displayed by the Configurator if the correct version number was found on the FC. 

Only new 'incompatible' sections that would need "tagging".  I'm thinking of #773 as an example here.

richard-scott, 2017-12-28:
Another example is someone trying to use 'ibus' on an FC version that doesn't seem to support it:
https://github.com/betaflight/betaflight/issues/4841

mikeller, 2017-12-28:
I don't quite understand this. How is what you are proposing different to the more than 300 places where we are checking the MSP API version in the existing app?

richard-scott, 2017-12-28:
ok, as an example if you connect an old FC and the BFC shows 'ibus' as a supported RX type, yet the FC firmware doesn't support 'ibus' as its too old, then that's what I'm talking about.

closed 2017-12-28 by richard-scott

basdelfos, 2017-12-28:
Yes that true, some features and receiver protocols are left out on F3 targets (some more then others) to make room for other features. The BFC doesn't check which features or rx protocols are (not) available in the flashed betaflight firmware. I think the rx protocol part (and the feature switches) are the only parts that are not checked as this is not version depended.  When we are moving to MSPv2 we should consider implementing checking for unavailable features.

richard-scott, 2017-12-28:
@basdelfos thank you for your support. @mikeller seems to have a very narrow vision of what is possible. I've seen a fair bit of "RotorRiot" attitude in his posts on other issues which doesn't help.

Please feel free to re-open this ticket (or start another one) if you need to.

mikeller, 2017-12-28:
@richard-scott: Please be a bit more careful with your statements. You open a ticket that asks very vaguely for something that we already have a process for, and are spending a considerable amount of time on, citing an example where exactly this process has already been applied. When I ask you to clarify what exactly it is you want us to improve in this process, you react by closing the issue, and then resort to accusing me of having a narrow vision and insulting me. This behaviour will only get you banned from GitHub.

For the next time, when you see something that you think needs improving:
1. try to read the code in the changes that have already done to fix what you see as wrong. Maybe there is already a process for fixing this, and you just don't know about it;
2. try to come up with concrete suggestions (code examples would be great too) of how you would like to fix what you see as wrong;
3. if what you came up in 2. accomplishes the same as what you saw in 1., then maybe your post should not be about generically fixing this perceived issue (since it already is fixed), but about concrete instances where this process still needs to be applied;
4. if 3. is the case, probably a better strategy than opening an issue to point out the problem is to fix it, and open a pull request for your fix.