betaflight/betaflight-configurator #828: Cannot run under ubuntu 17.10

wx4cb, 2018-01-02:
Hi all

just reinstalled the latest ubuntu on my laptop, everything went in ok, compiled fine.

I did an gulp release --linux64 and it compiled and packaged up a nice release archive.

Extracted it to a temp directory and tried running it from bash, all good. But when i tried to run it from the file manager "files", it says:

"There is no application installed for shared library files. do you want to search for an application to open this file"

everything seems ok with the file and as i can say, i can run it fine from the bash prompt, but i would love to be able to desktop a link. 

charlesb@HPLaptop:/usr/src/betaflight-configurator/release/Betaflight Configurator# file betaflight-configurator 
betaflight-configurator: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, stripped

basdelfos, 2018-01-03:
I've noticed the same behaviour on Ubuntu 16.04 LTS. Trying to run from the file manager doesn't work and raised the same error. Running from the terminal seems to work fine.

@mikeller you are a more Linux expect on this, can you give this a look?

richard-scott, 2018-01-03:
What about the binary file in the "apps" folder? Does that run?

Also, you can run "gulp apps", it won't build the zip, but it will give you an "apps" folder with what would have been the zip files contents to use. saves unzipping it.

Are you able to give me a command list of how to get Ubuntu setup for this to compile?
I have it compiling on Windows but never tried on Linux.

basdelfos, 2018-01-03:
@richard-scott install Ubuntu 16.04 LTS or 17.10. 
Once installed, run the following commands:

Install Node.js
```
curl -sL https://deb.nodesource.com/setup_8.x | sudo -E bash -
sudo apt-get install -y nodejs
```
Install git
```
sudo apt-get install git
```

Download the git repository and follow the steps in README as used under Windows (=platform independent).

wx4cb, 2018-01-03:
@richard-scott  i'll give that a try later

McGiverGim, 2018-01-03:
It seems that the problem is that the file is "detected" as "application/x-sharedlibrary", and not as executable when opened from the file manager.

It seems to be a known issue:
https://github.com/nwjs-community/nw-builder/issues/401

The solution is to add a `.desktop` and execute it.

McGiverGim, 2018-01-03:
I've added this `betaflight-configurator.desktop` and it worked (change the path to yours):
```
[Desktop Entry]
Name=Betaflight Configurator
Exec=/home/miguel/git/betaflight/betaflight-configurator/apps/betaflight-configurator/linux64/betaflight-configurator
Terminal=false
Type=Application
```

basdelfos, 2018-01-03:
Good find. Seems like that adding a `.desktop` file is the best solution.

AndersHoglund, 2018-01-03:
Re the advice to "...follow the steps in README..".... This page/file would need a major overhaul. Especially end user instructions how to install and use. Still refers to Chrome sore and running it as Chrome App. 
Also the development and building sections would need some refresh. Dependenciy on NSIS in Windows for example.

basdelfos, 2018-01-03:
@AndersHoglund there is a section in the README that tells how to build the standalone app. But true, the Chrome Store section needs to be removed and the NSIS has to be added.

McGiverGim, 2018-01-03:
This other `betaflight-configurator.desktop` does not need to specify the complete path, so a temporary solution can be add this file to the zip:
```
[Desktop Entry]
Name=Betaflight Configurator
Exec=sh -c "$(dirname %k)/betaflight-configurator"
Terminal=false
Type=Application
```

basdelfos, 2018-01-03:
Yes, that's what I though also. Will add the `.desktop` file to all the standalone apps later today.

wx4cb, 2018-01-03:
Sorry, im talking about the standalone app not the chrome app

So just rename it with .desktop and it should ve good

McGiverGim, 2018-01-03:
@wx4cb no, you must CREATE a new file, called `betaflight-configurator.desktop` with the contents of my latest message. Then, from the file manager, you must execute this new file.

wx4cb, 2018-01-03:
@McGiverGim that's what I thought, i was reading on my phone and it just didnt sound right :D

closed 2018-04-17 by wx4cb