betaflight/betaflight-configurator #834: Save and reboot issue

Gorilla-1983, 2018-01-07:
I'm having a problem with a Betaflight F4, I am trying to configure a serial port for ibus. When I hit save and reboot I get the following error:
2018-01-07 @ 15:43:15 -- EEPROM saved

2018-01-07 @ 15:43:16 -- Unrecoverable failure of serial connection, disconnecting...

2018-01-07 @ 15:43:16 -- Failed to close serial port

When I go back into the ports tab the changes I made haven't been saved.

Has anyone experienced this error before or know what I'm doing wrong?

mikeller, 2018-01-09:
Could be a corrupted configuration. Try resetting to defaults and re-applying your configuration.

richard-scott, 2018-01-09:
I had that happen with v10.0.0 of the Configurator, it was resolved for me by using the latest from GitHub (v10.0.1).

Gorilla-1983, 2018-01-09:
I realised that I was making an error with the configuration that was causing a conflict, therefore not saving changes. Changes are saved to the configuration when done correctly, however when I hit "save and reboot" the issue with the serial port occurs.
Where is the 10.0.1 Configurator located?

richard-scott, 2018-01-09:
@Gorilla-1983 Download and unzip [this](https://github.com/betaflight/betaflight-configurator/archive/master.zip).

Then use the "Load unpacked extension button" in the Extensions screen of Chrome.
You will then have two versions installed, the Chrome Store one, and this one.

You need the v10.0.1 one.

Gorilla-1983, 2018-01-09:
I attempted to "save and reboot" with the 10.0.1 Configurator and had the same error occur

Gorilla-1983, 2018-01-09:
When I exit the cli after making configuration changes I have this error;
2018-01-09 @ 23:12:19 -- Unrecoverable failure of serial connection, disconnecting...

2018-01-09 @ 23:12:19 -- Failed to close serial port

2018-01-09 @ 23:12:19 -- Serial port successfully opened with ID: 3

2018-01-09 @ 23:12:29 -- No configuration received within 10 seconds, communication failed

2018-01-09 @ 23:12:29 -- Serial port successfully closed

and the FC doesn't reboot

Gorilla-1983, 2018-01-09:
I attempted to "save and reboot" with 10.0.1 Configurator and had the same error occur.

Also when I make changes in the cli and type save I get the following;

2018-01-09 @ 23:10:36 -- CLI mode detected

2018-01-09 @ 23:12:19 -- Unrecoverable failure of serial connection, disconnecting...

2018-01-09 @ 23:12:19 -- Failed to close serial port

2018-01-09 @ 23:12:19 -- Serial port successfully opened with ID: 3

2018-01-09 @ 23:12:29 -- No configuration received within 10 seconds, communication failed

2018-01-09 @ 23:12:29 -- Serial port successfully closed

and the FC doesn't reboot

richard-scott, 2018-01-09:
Have you tried updating the serial drivers? Maybe this [video](https://www.youtube.com/watch?v=rJhJxc-Yn4o) will help.

sadfst, 2018-01-22:
Have you resolved your issue yet? I am having the same issue.

Gorilla-1983, 2018-01-26:
Hey sadfst and everyone else following this thread, I have an update to the issue I was having. It appears after a save and reboot, the serial port would open with a different ID to the Com port; in my post you can see the serial port after the reboot is serial 3 - my BetaFlight Configurator is set to Com11. The Configurator would then connect when I pressed connect.
The reason my changes to the ports tab weren't being saved was because I was creating a conflict.
Long story short, config changes are being saved but the reboot is not rebooting to the correct serial port.

andrew0007, 2018-02-10:
Hi, I have the same issue... can this have any side effect?
how to avoid that the reboot is not rebooting to the correct serial port?
Thanks a lot.

czchc, 2018-02-11:
I have the same issue with OMNIBUS F4, F7 and a Matek411.

andrew0007, 2018-02-11:
I am using Matek F405-CTR and I installed the driver with Zadig.

AndersHoglund, 2018-02-11:
Install the STM32 USB-VCP driver directly, Zadig is not really applicable in this case. 
All linked in the WiKi:  https://github.com/betaflight/betaflight/wiki/Installing-Betaflight#installing-stmicro-virtual-com-port-vcp-driver-under-windows

Christof-Eugster, 2018-03-10:
Ive got a OmnibusF4SD
Same Problem. Ive troubleshooted everything USB related.
After a fresh reflash with 3.3.0 i could once (1x) save and reboot with success, after that
i get this message "Unrecoverable failure of serial connection, disconnecting..."

The changes were saved, but to reconnect with betaflight i had to unplug/replug the usb cable.

When Downgrade to 3.2.5 no issues!

andrew00077, 2018-03-10:
Hello, 

I confirm that even updating the USB drivers to the latest versions I still have the same issue: when I save the Betaflight config disconnects (but does the save correctly). It happens with BF3.3 and I had same issue with BF2.4.

closed 2018-05-13 by stale[bot]

Christof-Eugster, 2018-05-25:
Hi John955
When i remember right it was because i loaded the wrong firmware, there were two of the same
but one for onboard sd card. In the manual for my f4 controller it was mentioned to take the one with
Memory option. Try to load up the other one... may it helps...

John955, 2018-05-29:
Thank you very much.  I got it to load.

John955, 2018-05-29:
Thanks and hi,
I had one more question.  When using a serial DSM2 orange extra reciver, I am able to see changes in my input, in the Betaflight, when I move my remote control sticks, but they are random and multiple channels change at a time, so I know it is not a channel mapping problem.  Any ideas on why the board is not reading my reciver input correctly?

Thanks,

John

Christof-Eugster, 2018-05-29:
May it is because of the different BF Firmware.
May you have to update the firmware of your receiver.
May it is only a binding problem between receiver and transmitter or a bad setting.
Or a unclean power flow. Or a bad solder....
I dont know your receiver, so...
After your post ive tried to load the newest betaflight OmnibusF4SD firmware again.
And suddenly it worked. It still doesent close the Port but the changes finally are saved.
It can cause problems when loading up the wrong Firmware... So may you now try to make a
fresh firmware update with the correct FM for your Flightcontroller.
This is a odd problem...
What i can say... i will never buy A GEPRC Stable controller again :)

AndersHoglund, 2018-05-29:
@John955 Typical symptoms when the wrong protocol SPEKTRUM1024 or SPEKTRUM2048 is selected. Will happen if Rx is bound with 1024 and BF is set for 2048, or the other way around.
Q: Why is this reported/asked in a stale BFC case dealing with Save & Reboot? Very OT and more a support question than a BFC issue. GitHub issues are intended for reporting/discussing software problems, not support cases.

Christof-Eugster, 2018-05-29:
Funny enought just right now ive flashed a new firmware for a different model and the exact same happend to me.
It was as AndersHoglund says. For some reason Betaflight decided to change sbus to spektrum1024... lol

ylectron, 2019-01-02:
I just have the same issue and I think I know where this is coming from. 
I had to make up a soft serial because ..reasons, and the f.port receiver setting would not stick until the rx pin was also defined. I did define it on my PPM pin, and everything seems to be fine except that BF let me know that that particular resource A00 in my case was also assigned to ESCSERIAL... probably the serial pass thru for the esc programming. I am convinced that this is what prevents the other serial port from closing.

jasenhowell, 2019-01-21:
hi i am using betaflight configurator 10.4.1 and flashing 3.5.5 OMNINXT7 15-01-2019 firmware to the  LHI OmniNXT F7 Flight Controller FPV FC Upgrade Version of Omnibus F7 V2 board.i have tried other releases and  every time i hit save and reboot i get this same error
i have also tried INAV and get the same issue
2019-01-20 @ 12:40:35 -- Serial port successfully opened with ID: 31
2019-01-20 @ 12:40:35 -- MultiWii API version: 1.40.0
2019-01-20 @ 12:40:35 -- Flight controller info, identifier: BTFL, version: 3.5.5
2019-01-20 @ 12:40:35 -- Running firmware released on: Jan 15 2019 10:17:07
2019-01-20 @ 12:40:35 -- Board: ONX7, version: 1
2019-01-20 @ 12:40:35 -- Unique device ID: 0x33001e3137510f30343533
2019-01-20 @ 12:40:35 -- Craft name:
2019-01-20 @ 12:40:35 -- Arming Disabled
2019-01-20 @ 12:40:45 -- EEPROM saved
2019-01-20 @ 12:40:45 -- Unrecoverable failure of serial connection, disconnecting...
2019-01-20 @ 12:40:46 -- Failed to close serial port
does anyone have any solutions?

DaViDeS0LaZz0, 2020-07-19:
I am experiencing the same issue.
How do you fix it?
Thx

wmirza89, 2020-07-23:
Hello all - I am experiencing the same issue. Anyone have any solutions?

brandonhill09, 2020-09-18:
Hello, I am trying to configurate my practice drone as I am just getting into drone piloting. I got it to work at first, but after a few changes I got a "Failed to close serial port" error after I tried to save and reboot my drone. Please tell me how to fix this? I really want to fly again :D

haslinghuis, 2020-09-18:
Did you try the driver fix on https://impulserc.com/pages/downloads

brandonhill09, 2020-09-18:
> Did you try the driver fix on https://impulserc.com/pages/downloads

No I haven't, how do I do that?

haslinghuis, 2020-09-18:
Google it?
https://www.youtube.com/watch?v=9yQYVfHgmD8

brandonhill09, 2020-09-18:
It didn't work. My changes still aren't saved, and the same message still comes up despite the fix.

haslinghuis, 2020-09-18:
Did you install this one: https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

brandonhill09, 2020-09-18:
No, what does it do?

brandonhill09, 2020-09-18:
Also, I installed it but I don't know how to add it to the betaflight application.

haslinghuis, 2020-09-18:
You don't. Drivers are an underlying layer used by the application. Also try a decent data cable.

brandonhill09, 2020-09-18:
Ok, will do. However, what if that doesn't work?