betaflight/betaflight-configurator #843: Problem building BFC on MacOS Sierra.

AndersHoglund, 2018-01-12:
Seems to be some strange dependencies not met, or circular deps rather, when building on MacOSx Sierra. I can make it work with this sequence, on a clean pure virgin git clone.

```
npm install gulp
npm install
gulp release --osx64            --> fails with "gulp.zip" missing
npm install gulp
gulp release --osx64            --> fails with "gulp-appdmg" module missing
npm install
gulp release --osx64            --> OK.

```
Prior to all this "sudo npm install --global gulp-cli" was made. 
Using macOS Sierra version 10.12.6 (There seems to be an update pending, will take that and retry.)

AndersHoglund, 2018-01-12:
Updated to macOS High Sierra. Same issues.
But I think I have found a work-around. This sequence seems to work on macOS:

```
npm install
npm install gulp gulp-appdmg
gulp release --osx64

```
Works also in Jenkins: http://andwho.sytes.net:8080/job/BorisB_BetaFlight_Configurator/

AndersHoglund, 2018-01-12:
Hmm, I did get this, once on Jenkins and once in an interactive shell:
```
+ gulp release --osx64
[18:41:19] Using gulpfile /private/var/jenkins/workspace/BorisB_BetaFlight_Configurator/Platform/MacOSx/gulpfile.js
[18:41:19] Starting 'clean-dist'...
[18:41:19] Starting 'clean-apps'...
[18:41:19] Starting 'clean-release'...
[18:41:20] Finished 'clean-release' after 149 ms
[18:41:20] 'clean-apps' errored after 984 ms
[18:41:20] Error: EINVAL: invalid argument, rmdir '/private/var/jenkins/workspace/BorisB_BetaFlight_Configurator/Platform/MacOSx/apps/betaflight-configurator/osx64/betaflight-configurator.app/Contents/Resources/app.nw/node_modules/appdmg/test/assets'
[18:41:20] Finished 'clean-dist' after 991 ms
Build step 'Execute shell' marked build as failure
```
This was copied from the Jenkins log. But "one is none", next build was fine. Let's just keep it running like this and see what happens.

mikeller, 2018-01-13:
Try updating to latest, and following the steps in #845.

AndersHoglund, 2018-01-13:
Thank you, works much better now after pulling current master. Builds OK for OSx on both Jenkins and in an interactive shell, no more npm special case needed. 
But `npm install` still creates a modified `package.json`. Seems to be no real problem when building incrementally. But anyhow, why not commit this, so initial first run has the same info as the second etc?
But this change is a dependency specific for OSx only, might not work on Linux and Windows. Or?
```
$ git diff
diff --git a/package.json b/package.json
index d43cc144..f265dc00 100644
--- a/package.json
+++ b/package.json
@@ -21,7 +21,9 @@
   },
   "author": "The Betaflight open source project.",
   "license": "GPL-3.0",
-  "dependencies": {},
+  "dependencies": {
+    "gulp-appdmg": "^1.0.3"
+  },
   "devDependencies": {
     "del": "^3.0.0",
     "gulp": "~3.9.1",

```

mikeller, 2018-01-13:
Hmm, that's odd. I never noticed this happening when building on MacOS. (But I also never specifically checked for it.)

The reason we are installing `gulp-appdmg` by means of `platform-dependent-modules` is that it only exists as a MacOS package - including it as a dependency (or as a devDependency which is more appropriate) will make `npm install` fail on non-MacOS.

AndersHoglund, 2018-01-13:
No surprice, but exactly the same thing happens to `package.json` in BelHeliConfigurator and BlackBoxExplorer builds. At least a consistent behaviour. And as it seems to do no harm when building, I will close this issue. It could be a annoying thing when/if developing on OSx and (inadvertently) committing this self-modification,  possibly causing problems on other platforms. 
Are there no way to have platform specific deps in npm package handling? Never mind, retoric Q.

closed 2018-01-13 by AndersHoglund