betaflight/betaflight-configurator #859: Translators wanted

mikeller, 2018-01-16:
If you'd like to help us in our effort to make the configurator available in more languages, please reply to this post with the language(s) you can help in, and we'll post instructions on how to get going.

RipperDrone, 2018-01-18:
German :-) Happy to help

McGiverGim, 2018-01-18:
Thanks @RipperDrone you only need to go to http://betaflight.oneskyapp.com and collaborate with the German translation 😁

RipperDrone, 2018-01-18:
consider it done :-)

SwiftingSpeed, 2018-01-19:
I could maybe help with Swedish

SwiftingSpeed, 2018-01-19:
Yea I saw that now, it's going to be hard doing it by myself. I'll ask other swedes to help.

RipperDrone, 2018-01-19:
Completed checking German, all 21 pages. Done quite some changes, hope they are good :-)

tonymannaro, 2018-01-19:
hi, I would be happy to help for **ITALIAN**
ciao!

RipperDrone, 2018-01-19:
Nice - so great to see a community of helping hands here. It's a great way we can give something back to the dev team / project :-)

tonymannaro, 2018-01-19:
i've  gone here http://betaflight.oneskyapp.com but i didnt find italian.... i'm missing something?

McGiverGim, 2018-01-19:
@tonymannaro Italian added, you can start when you want 🤗

RipperDrone, 2018-01-19:
General question: Can language version be selected in BFC? Would like to switch btw EN and DE to check things out, particularly regarding spacings etc. (DE seems more spacy, some buttons flow over...)

McGiverGim, 2018-01-19:
@RipperDrone Yes, in the text of https://github.com/betaflight/betaflight-configurator is there a paragraph called languages, with instructions.
And if you used the windows installer or the Debian installer, you have two shortcuts in all programs menu, one for English and other for system language.

abacsSk, 2018-01-23:
i can translate to Hungarian and Slovak languages.

McGiverGim, 2018-01-23:
@abacsSk please tell me what do you prefer to start with. When you finish the first, I can add the other if you want ;)

abacsSk, 2018-01-23:
I prefer hungarian. It will definitely take a few days to finish it...

McGiverGim, 2018-01-23:
@abacsSk Hungarian added. Thanks!

piodabro, 2018-01-23:
Please add Polish

McGiverGim, 2018-01-23:
@piodabro added!

feiticeir0, 2018-01-24:
I can help with European Portuguese !!

McGiverGim, 2018-01-24:
@feiticeir0 I suppose you're talking about portuguese from Portugal (pt_PT). If that's correct, added ;) You can start when you want.

McGiverGim, 2018-01-24:
One thing more to all translators: until now, the translations were auto-approved. Since now the translations will not be "auto-approved".
You can continue working, but the strings will not be approved until one of this things happen:
- A moderator of a your language approve it.
- The translation has 3 votes (this number can be changed in a future, we will see).
- For the languages that haven't a moderator, I will "auto-approve" the phrases with most votes from time to time. 
Finally, I wold like to have two moderators by language, so if someone is interested in being one, please go to the slack chat of betaflight and send me a direct message. Thanks!

tonymannaro, 2018-01-24:
guys, while traslating italian, i think (not sure ) to have found a typo in the english text:
#39231314
increase the value **_or_** to gain more idle speed.
maybe im wrong but i suppose that "or" is not in th eright place :)

McGiverGim, 2018-01-24:
Yes, pretty sure that is a typo :) 
The right solution, until now, will be to make a PR to solve this, but a user with "developer" level is needed. Another solution is to change it in the translators portal, english language, but a merge with the new phrases added in code will be needed in a future, and I suspect that manual... 

@mikeller some idea about this?

tonymannaro, 2018-01-24:
"SPI Bus Receiver Provider" is identical in  
#40135306 and #39233231

McGiverGim, 2018-01-24:
Yeah, one of them will disappear because has been changed in origin. My fault to not hide the other. I will do it now. Thanks!

tonymannaro, 2018-01-24:
same issue for #39231380 and #39231410 "save and reboot"

tonymannaro, 2018-01-24:
#39233249 and #39233243 
if im boring with theese.... please tell me  :)

tonymannaro, 2018-01-24:
#39231608
#39231680
#39231701
#39231830
#39231896
#39231938
#39232304

all have the same phrase.

tonymannaro, 2018-01-24:
#39231470
..."but this functionality **_requires requires_**"...

feiticeir0, 2018-01-24:
@McGiverGim Yes, that is it ! Thank you ! Already found the files and will start soon !

McGiverGim, 2018-01-24:
Sometimes the phrase is the same, but they are in two different places. Sometimes the translations can be shared, but sometimes is better to have two translations, maybe in a future some of them change, or in english the phrase is the same but not in some of the translation (strange, but can occur).

tonymannaro, 2018-01-24:
ok... thanks ..anyway i have traslated them all at the same way for the moment, but at the end i will try to do some check side by side in the application, so i can contextualize avery phrase. Effectively it can occur... you are right, sorry i dont considered it before.

McGiverGim, 2018-01-24:
I'm pretty sure that this will not be the case, and all the phrases will be the same 😊 I will build this week a version with all the languages if someone want to test them. I'm waiting for some PRs to do it.

Barnabasek, 2018-01-24:
I can help with Czech translation.

McGiverGim, 2018-01-24:
@Barnabasek Czech added 😉

McGiverGim, 2018-01-25:
Uploaded master with all the translations up to date, and the menu to change language, to let test the translations https://drive.google.com/drive/folders/1w-mtfc0xV-IN62YLVxJ0kOEaJ8u0wbSV?usp=sharing

The ' hot language change'  need some work more, so don't add issues about this. This build is only to test the work done in the language translation.

tonymannaro, 2018-01-25:
great! thanks!

tonymannaro, 2018-01-25:
..it is blocked by windows smart screen  :(

McGiverGim, 2018-01-25:
@tonymannaro I've uploaded a different build, please test if this helps. If not, I've uploaded a `portable` version too ;)

tonymannaro, 2018-01-25:
Many thanks
I can't try it until tomorrow I think unfortunately... Sorry.  I will let you know as soon as I can.

tonymannaro, 2018-01-26:
@McGiverGim i have tried both new installer and portable ... but in both case i receive the same error message by smart screen ...and the "wonderful" thing is that i have only one button "dont start".  i think i should disable microsoft defender smart screen but i cannot at the moment on this pc. i hope it is affecting me only. :(

McGiverGim, 2018-01-26:
So this problem does not happen to you when installing, only when executing?

tonymannaro, 2018-01-26:
both case.. every time i try to launch the executable file by you it seems :) ...just for sure ...im going to try some other thing

tonymannaro, 2018-01-26:
ok... tried with some other executable (winrar installer and a driver installer and a phothoshop portable exe file) all of them worked fine

McGiverGim, 2018-01-26:
The smartscreen validates agains "known" executable files, where they were downloaded, etc. The winrar and photoshop are mainstream, so they will not throw any error. The new "betaflight-configurator" created is not known by microsoft, and they will block it. 
I don't know what rules they use, if I can find a workaround a will let you know.

isilcala, 2018-01-30:
Simplified Chinese, happy to help.

Elvin

McGiverGim, 2018-01-30:
@isilcala there's one user translating the Simplified Chinese, and the work for the first version is almost ready, but help is always welcome.

isilcala, 2018-01-30:
Cool. Let me know whenever needed ;)

McGiverGim, 2018-01-30:
You can start now, there're some new strings that were not translated ;)  http://betaflight.oneskyapp.com

feiticeir0, 2018-01-30:
Having a problem with a translation.

without the context, can't decide what's best:

Tethered logging.. I know what's Tethered, and now what's logging, but
without the context, can't decide.. Where is this string in betaflight ?
thank you !

McGiverGim, 2018-01-30:
Is one of the principal tabs. In advanced mode, you will see it in the left, with the others, like blackbox, etc.

isilcala, 2018-01-31:
@McGiverGim Consider Simplified Chinese done ;)

McGiverGim, 2018-01-31:
Perfect! I will talk with the other guy and push the PR today. Thanks for the work!!

Saxin, 2018-01-31:
This was already requested few times I think...
Can we have lang. switch dialog - say in the gears icon (upper right) ?

Saxin, 2018-01-31:
@McGiverGim  sorry for last post just saw that lang. switch dialog was added to master, I would like to help with Hebrew lang. [Hebrew he he 1037 1255]

McGiverGim, 2018-01-31:
Don't worry. Hebrew added. You can start when you want. 😁

flash-v, 2018-02-01:
I can help with Russian.

McGiverGim, 2018-02-01:
@flash-v Russian added. Thanks for your help! You can start when you want http://betaflight.oneskyapp.com

jonathanholt, 2018-02-02:
I can help with Japanese. I have a Japanese language proficiency certificate and usually collaborate with native friends to check over my translations.

McGiverGim, 2018-02-02:
@jonathanholt Japanese added. You can start when you want http://betaflight.oneskyapp.com

Thanks for the work!

[72 more replies]