betaflight/betaflight-configurator #864: Win10 sometimes unreadible fonts - looks like a 'clearfont' issue?

RipperDrone, 2018-01-16:
[image]

RipperDrone, 2018-01-16:
Fonts scaling seems to cause dithered / unreadible fonts sometimes...

McGiverGim, 2018-01-17:
I'm not too sure that this is a bug. The configurator uses "web fonts", when the connection is slow, or it is not a connection at all, it will fall back to a font with less quality. Maybe this is the problem. When this happens, do you have a good internet connection?

RipperDrone, 2018-01-17:
would think so, DSL16k...

basdelfos, 2018-01-18:
May be this has something to do with the global font size in Windows. Can you tell us what your font size is?

[image]

Sorry...Dutch locale on this laptop, but the menu items are in the same place.

RipperDrone, 2018-01-18:
It’s @ 100% on 1920x1080 res. Is this what you were asking? :)

 

 <https://user-images.githubusercontent.com/334779/35086612-6510a23e-fc2d-11e7-9494-91e99a2807b6.png>

basdelfos, 2018-01-19:
Yes, that's what I meant. Strange, nothing unusual with those settings.
Now I look at your image again, it looks like that the whole inner part of the app is blurry and not only the fonts?

McGiverGim, 2018-01-19:
Yes, the images are blurry too (the `donate` button for example...) Maybe a problem with the graphic card?

basdelfos, 2018-01-19:
I don't think it is hardware related, I think it is chromium (part of nw.js) releated. 
Do you have the same issue with the standalone Blackbox Explorer?

RipperDrone, 2018-01-19:
No, BBE is crystal clear: 
[image]

RipperDrone, 2018-01-19:
Comparison: 
[image]

RipperDrone, 2018-01-19:
When pressing some buttons (minimizing/maximizing app window, it looks normal again... <noideawtfisgoingonhere>

RipperDrone, 2018-01-20:
crazy... 
[image]

RipperDrone, 2019-10-20:
resolved - NVidia Antialiasing (FXAA) needs to be switched off

closed 2019-10-20 by RipperDrone