betaflight/betaflight-configurator #867: Build configurator for Raspbian on Raspberry pi

lalek54, 2018-01-17:
Hi 
I me trying to buil the configurator app for my Raspberry Pi Raspbian but my app (—linux32) is build for i386 processor. I don t know how to build it for armv7 processor.

mikeller, 2018-01-17:
Hi @lalek54.
The NW.js toolkit that is used to build the configurator standalone version currently does not support the arm architecture. There is some discussion around this, and hints on how to manually build NW.js on arm here: nwjs/nw.js#193
Feel free to have a go at this, and report back, there might be others that would like to get this to work.

lalek54, 2018-01-17:
Thx a lot.
So i tried to launch the configurator with the build of nw.js for raspi but i have a lot of errors related to gdkpixbuf.

lalek54, 2018-01-18:
I managed to build a Raspbian version for raspi3. 
I need to make sure everything is working proerly and then i will try to package it to share.

mikeller, 2018-01-18:
Good going, well done!

AndersHoglund, 2018-01-18:
Good Job! Did you build on RaspPian? Is it possible to crosscompile for ARM and Raspian on a x86 Linux host?

lalek54, 2018-01-18:
No need to compile. It s à Node webkit project. 
So the only thing we need is a version of NW.js which is compiled for armv7. You can find it there 
https://github.com/LeonardLaszlo/nw.js-armv7-binaries/releases/
Then You add all the files of nw.js in the unziped folder of your betaflight configurator directory. 
Then You can launch the app by typing ./nw
But i will try to provide a zip file with a clickable icon soon.

AndersHoglund, 2018-01-18:
OK, great. If it is only a matter of downloading and packaging then it could be done by Jenkins (running on x86 Linux) . Still, this is a form of "building" by some scripting, even if not compiling. Would you mind publishing the script? 
Current Configurator build here: http://andwho.sytes.net:8080/job/BorisB_BetaFlight_Configurator/  

We could do Raspian version of BB-Explorer and BlHeli-Configurator too perhaps.....

AndersHoglund, 2018-01-18:
Maybe Raspbian support could be added to the `gulpfile.js` in a similar way as `chromeos` support is done? I.e. wget the required tar-ball, extracting the nw.js files and repackage in a way suitable for Raspbian installation.

lalek54, 2018-01-18:
I don t have a script at thé moment. I me totally beginner whith Linux, nw.js, gulp and so on.
I did it manually. 
I have a .desktop file wich is clickable now and i can launch the configurator easily. 
Now i need to investigate the gulpfile.js to understand how the chromeos app is build and to try to duplicate the process. 
If somebody can éducate me on gulp it could be easier.

lalek54, 2018-01-18:
So it s Easy to clone the process of building linix32 app on gulp. 
I just replaced the files in the cache directory with the arm version. 
And then Install Node and « gulp apps —linux32 » 
It s automatically building a working app for raspbian armv7. 
If can tell me how to édit thé gulpfile.js to add à new platform?

McGiverGim, 2018-01-19:
The problem is that the cache folder is downloaded automatically by the builder. It's not so easy to modify it.

lalek54, 2018-01-19:
Here is my app. 
https://www.dropbox.com/s/5yw5nu7kz37quoi/betaflight-configurator10.1.0.zip?dl=0

lalek54, 2018-01-19:
Yes i understand that. I need to add a platform to gulpfile.js (e.g: Linux/arm). 
Clone the building process of Linux 32 app. Modifie this process to make it download and extract this zip file 
https://github.com/LeonardLaszlo/nw.js-armv7-binaries/releases/download/v0.23.7/nwjs-sdk-v0.23.7-linux-arm.tar.gz

McGiverGim, 2018-01-19:
At least you must:
- Add `arm` platform to the method that reads the input parameters
- Remove this parameter when calling the `nwbuilder` (like `chromeos` does) and add `linux32` if it isn't
- In the post-build method, when `arm` is a platform,, copy the `linux32` folder from `apps` to a `arm` folder inside `apps`, download the tar.gz, and decompress/rename the contents.
- Then the release method for arm must call only the `archive_zip` method.

Another solution, that you can test is to use:
- https://www.npmjs.com/package/nw-with-arm but I'm not too sure if this is a good idea, it replaces the actual nw builder, but who knows...

McGiverGim, 2018-05-12:
The app folder is being generated automatically when you run the gulp process. It does not exist until that moment.

nmaggioni, 2018-05-12:
Right, I was working to get debug task working first and didn't notice that.

QuadFramesUK, 2018-06-06:
I have just finished programming my Raspberry Pi 3 B with Raspian OS to run any version of Betaflight, Butterflight & Blheli. 

Just uploading the how to video for the betaflight install now on YouTube. 

if you click the link to my channel, hit subscribe and click the bell you will get a notification / email when Butterflight & Blheli vids are uploaded.

https://www.youtube.com/c/QuadFramesUK

QuadFramesUK, 2018-06-07:
https://youtu.be/dDTcTppLfqk

closed 2018-06-09 by mikeller